autosportlabs/RaceCapture-Pro_firmware #545: Implement Extended Kalman Filter for GPS/IMU Sensor Fusion

kosenko-max, 2016-05-02:
As far as I can see from the code GPS data now comes unfiltered and thus contains a lot of errors and data drop outs. 7 axis of accelerometer, gyroscope and magnetometer do some kind of sensor fusion already to improve output (may be not to the level of Bosch 9 axis). Standard approach in the drones software is to fuse all those sensor signals in Extended Kalman Filter which takes into account vectors of predictions and assumes probability of one or another movement based on inertial model. 

This allows to achieve very high statistically optimal precision of positional data with higher rates than GPS alone. Plus improves acceleration and rotation reporting.

Lower logging rate should be used to improve accuracy while using same calculations and denoising instead of just losing some data. This is already done on chip, but with sensor fusion it produces better results if data is taken at maximum speed.

This is not a small task, but there are plenty of libs and knowledge available nowadays publicly.

stieg, 2016-05-02:
Not a bad idea.  Any links to existing C code or something along those lines would be useful.

kosenko-max, 2016-05-02:
There are some libraries for embedded implementations like this one https://github.com/dr-duplo/eekf
Some are without GPS like that one https://github.com/TKJElectronics/KalmanFilter
Or this one https://github.com/simondlevy/TinyEKF

brentpicasso, 2016-05-02:
Thanks, Max for the links. we've talked a lot about doing sensor fusion with IMU + GPS to enhance the GPS system. Good stuff.