autosportlabs/RaceCapture-Pro_firmware #548: Implement txCANQueueFullTime(channel) method in Lua

kosenko-max, 2016-05-02:
For non-blocking txCAN implementation it makes sense to make this function that checks if queue is full and there is no reason to do any request for now. It returns 0 if queue is not full and number of milliseconds it stays full (to identify problem with CAN connection). This way Lua script would not be wasting time preparing and transferring request data again and again.

stieg, 2016-05-02:
> For non-blocking txCAN implementation it makes sense to make this function that checks if queue is full and there is no reason to do any request for now

Hrmm... interesting request.  Well because of issue #549 you can't send CANs using async methodology until 2.9.1 goes out the door.  But if you specify a timeout of 1ms you are pretty close to async (but of course... not quite there yet).

Help me better understand what you are trying to do that would necessitate something like this.  From my perspective I would like to keep things as dirt simple as possible.  Perhaps from your needs we can figure out a good solution (also no pun intended).

kosenko-max, 2016-05-02:
Ok, I'm trying to do fuel pumps management together with logging (I know it sounds crazy, but it's only temporary solution and I can see that any other real time management of the equipment would not like to be stalled).

So I need to react quickly to any pressure changes and get as much data as possible to properly denoise it. And my can logging is request-response based which does something like 500 txCAN per second and 1000 rxCAN per second. But from time to time requests or responses are delayed. Which is fine for pump management except one thing - thread is stuck for way more than it should be.

Now looking at the code I understand that txCAN with 0 timeout is not async. There are still some reasons it won't deliver a message and throw it out. I think it makes sense to be able to read queue status (like number of empty slots in the buffer) and let Lua developer cancel outstanding messages one by one. CAN mailbox would need to be keep trying to send messages and if queue is full it should return false on txCAN immediately.

1ms timeout is really not going to make it. Because it will cancel a message if it's not gone in 1ms. And even if not - for my case it could end up in 20ms waiting time and drop to 50 samples per second of sensor data.

I'm completely fine it takes some time to send a CAN message. I just would like to send request, do something else and check back that queue is emptied and I can start waiting for answers from the CAN. So original proposal in this ticket would not work, but those methods will work:

 - bool canceledActually = cancelTxCAN(chanel, numberOfLastRequestsToCancel) // requesting more than available will clean them all.
 - int numberOfRequestsInQueue, int millisecondsRetrying = getCANQueueStatus(channel)

I don't mean exact syntax, but just to give an idea...

kosenko-max, 2016-05-02:
Referring #549 - my suggestion is:

- change txCAN to return messageID or false if queue is full
- string status = getTxCANMessageStatus(messageID)
- bool isOk = retryTxCANMessage(messageID)
- bool messagesCancelled = cancelTxCANQueue(channel) 
- int numberOfRequestsInQueue = getTxCANQueueSize(channel)

stieg, 2016-05-05:
Will evaluate if this is something that can be part of 2.10.0 but no promises.  Patches are always welcome.

kosenko-max, 2016-05-05:
I still can't make devenv to work and don't have bench board. And doubt you
would like blindfolded made PR.
But I don't see now how I can use 0 timeout on txCAN without any management
of results, so no async CAN?

Will evaluate if this is something that can be part of 2.10.0 but no
promises. Patches are always welcome.

stieg, 2016-05-05:
> I still can't make devenv to work

Did you follow the instructions at https://github.com/autosportlabs/RaceCapture-Pro_firmware/blob/master/README.md ?  If so... where are you stuck?

>  and don't have bench board

What do you mean by this?  One with debug pins?  If you have an RCP, you can do development with it.  Just need a working dev environment.  Its how I got started in this project.

> But I don't see now how I can use 0 timeout on txCAN without any management of results, so no async CAN?

Because its a new feature, it won't go into the 2.9.x series of firmware.  Earliest would be 2.10.x series.  And primarily my dev time is focused on bringing up and stabilizing RaceCapture so hence why I make no promises.  Plus what you mentioned above is probably much more than needs to be done.  Here is what I would suggest we do:

* modify `txCAN` to return a tuple: First param is whether or not message was queued, second param is the mailbox id.
* create `txCAN_status`.  This command would take a mailbox ID and return one of 3 statuses: PENDING, SUCCESS, FAILED.  Probably return this as < 0, 0, > 0 respectively.
* create `txCAN_cancel`.  This command would take a mailbox ID and return true if it was successfully canceled, or false if the mailbox was empty.

This would play quite nicely with the hardware as all of those methods exist already under the hood.  All that would need to happen would be to write up the lua bindings and add the device methods to our HAL to support the `status` and `cancel` methods.

stieg, 2016-07-14:
Bumping to 2.11.0 since we need to get 2.10 out and this is not critical.