autosportlabs/RaceCapture-Pro_firmware #558: Mk2 Won't build on Vagrant Ubuntu/Xenial

meikeric, 2016-05-03:
Unable to get firmware to build.
**Environment:** Windows 10, Vagrant, VirtualBox
**Vagrant:** Ubuntu:Xenial

I had to change the version of libcppunit from "libcppunit-1.13-0" to "libcppunit-1.13-0v5" in bin/vagrant_setup.sh as the first one is not available for Ubunut/Xenial.

**History**

> ubuntu@ubuntu-xenial:~/RaceCapture-Pro_firmware$ history
    1  dir
    2  cd ~
    3  ls
    4  wget https://s3-us-west-2.amazonaws.com/asl-firmware/vagrant_setup/gcc-arm-none-eabi-4_7-2013q1.tgz
    5  tar -xzvf gcc-arm-none-eabi-4_7-2013q1.tgz
    6  pico ~/.profile
    7  dir
    8  export PATH="gcc-arm-none-eabi-4_7-2013q1/bin:$PATH"
    9  source .bashrc
   10  sudo pip install crcmod https://s3-us-west-2.amazonaws.com/asl-firmware/vagrant_setup/asl_f4_loader-0.0.6.tgz
   11  sudo apt-get install python-setuptools python-dev build-essential.
   12  git
   13  git clone https://github.com/autosportlabs/RaceCapture-Pro_firmware.git
   14  ls
   15  cd RaceCapture-Pro_firmware/
   16  ls
   17  cd lib/lua/src/
   18  make PLAT=stm32_generic
   19  make PLAT=stm32
   20  make PLAT=stm32_generic
   21  make PLAT=stm32 generic
   22  cd ..
   23  ls
   24  cd ..
   25  l
   26  ls
   27  cd lib/lua/src/
   28  make PLAT=stm32 generic
   29* Makefile:109: recipe for target 'generic' f
   30  make PLAT=stm32
   31  make: *** No rule to make target 'stm32', needed by 'default'.  Stop.
   32  make
   33  make PLAT=stm32
   34  ./configure
   35  configure
   36  ls
   37  cd ..
   38  ls
   39  make PLAT=stm32 generic
   40  make PLAT=stm32_generic
   41  pico Makefile
   42  make
   43  make PLAT=stm32
   44  cd ..
   45  cd .
   46  cd ..
   47  ls
   48  cd RaceCapture-Pro_firmware/
   49  ls
   50  make PLAT=stm32
   51  cd bin/
   52  ./vagrant_setup.sh
   53  ls
   54  ./setup_asl_f4_loader.sh
   55  ls
   56  ./style
   57  ./style.sh
   58  ./vagrant_setup.sh
   59  pico vagrant_setup.sh
   60  ls
   61  sudo apt-get install libcppunit-1.13-0
   62  sudo apt-get install libcppunit-1.13-0v5
   63  pico vagrant_setup.sh
   64  ./vagrant_setup.sh
   65  pico vagrant_setup.sh
   66  ./vagrant_setup.sh
   67  cd ..
   68  make
   69  make PLAT="stm32"
   70  make PLAT=stm32 generic
   71  make PLAT=stm32
   72  cd platform/mk2
   73  make
   74  make PLAT=stm32
   75  ls
   76  pico config.mk
   77  cd ..
   78  cd rct/
   79  pico Makefile
   80  cd ..
   81  pico mk2/Makefile
   82  pico mk2/freertos.mk
   83  pico mk2/config.mk
   84  cd ..
   85  make
   86  git status
   87  pico bin/vagrant_setup.sh
   88  history

**Make Output**
> 
OK (130)
make[3]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware/test'
make[2]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware'
make[1]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware'
make mk2-package
make[1]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware'
make -C lib/lua PLAT=stm32 clean
make[2]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua'
cd src && make clean
make[3]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua/src'
rm -f liblua.a lapi.o lcode.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o                                                                                                  lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.                                                                                                 o lzio.o lrotable.o lauxlib.o lbaselib.o ldblib.o lmathlib.o loslib.o ltablib.o                                                                                                  lstrlib.o loadlib.o linit.o bit.o lua.o luac.o print.o
make[3]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua/src'
make[2]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua'
make lua-build
make[2]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware'
make -C lib/lua PLAT=stm32 generic
make[3]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua'
cd src && make generic
make[4]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua/src'
make all MYCFLAGS=
make[5]: Entering directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua/src'
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -ggdb -Os -Wall  -DLUA_OPTIMIZE_MEMORY                                                                                                 =0 -DLUA_USE_MKSTEMP=1   -c -o lapi.o lapi.c
make[5]: arm-none-eabi-gcc: Command not found
<builtin>: recipe for target 'lapi.o' failed
make[5]: *** [lapi.o] Error 127
make[5]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua/src'
Makefile:109: recipe for target 'generic' failed
make[4]: *** [generic] Error 2
make[4]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua/src'
Makefile:56: recipe for target 'generic' failed
make[3]: *** [generic] Error 2
make[3]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware/lib/lua'
Makefile:99: recipe for target 'lua-build' failed
make[2]: *** [lua-build] Error 2
make[2]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware'
Makefile:107: recipe for target 'lua-pristine' failed
make[1]: *** [lua-pristine] Error 2
make[1]: Leaving directory '/home/ubuntu/RaceCapture-Pro_firmware'
Makefile:175: recipe for target 'package' failed
make: *** [package] Error 2
ubuntu@ubuntu-xenial:~/RaceCapture-Pro_firmware$

stieg, 2016-05-03:
Here is your bug: `make[5]: arm-none-eabi-gcc: Command not found`  You are missing the toolchain in your PATH.  Print your path and check if the needed entry is there.

Looking at the command history in command 6 you used pico to (presumably) edit ~/.profile to add the MK2 toolchain to the path.  Then in command 9 you source ~/.bashrc, which is not the same as ~/.profile.  Probably part of (if not the entirety of) the issue.

meikeric, 2016-05-03:
Working now. Thanks.
I just checked the vagrant_setup script, and it does on export of the path, but for some reason, it didn't actually put it in the path. Any ideas?

stieg, 2016-05-04:
I assume you are referring to this:

```sh
cat <<EOF | setup_profile_file "asl_mk2_tc.sh"                                                                                                                                  
# ASL script to setup our bin path for MK2                                                                                                                                      
export PATH="${VAGRANT_HOME}/gcc-arm-none-eabi-4_9-2015q3/bin:\$PATH"                                                                                                                       
EOF   
```
In that script it places a file with those contents in the profile.d section so that when you login you will get those environment parameters.  It doesn't source it during its setup phase.

closed 2016-05-04 by stieg