autosportlabs/RaceCapture-Pro_firmware #616: "retrieving WiFi info" occuring at high rate

brentpicasso, 2016-05-14:
MK2 with r/2.10.0 sees this activity at a high rate in the logs - basically happening as fast as serial can process:

[ESP8266 Driver] Retrieving Wifi IP info
[serial (Aux)] TX: AT+CIFSR(\r)(\n)
[serial (Aux)] RX: 192.168.4.1(\r)(\n)
[serial (Aux)] RX: (\r)(\n)
[serial (Aux)] RX: OK(\r)(\n)
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Checking WiFi AP
[ESP8266 Driver] Retrieving Wifi IP info
[serial (Aux)] TX: AT+CIFSR(\r)(\n)
[serial (Aux)] RX: 192.168.4.1(\r)(\n)
[serial (Aux)] RX: (\r)(\n)
[serial (Aux)] RX: OK(\r)(\n)
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Checking WiFi AP
[ESP8266 Driver] Retrieving Wifi IP info
[serial (Aux)] TX: AT+CIFSR(\r)(\n)
[serial (Aux)] RX: 192.168.4.1(\r)(\n)

brentpicasso, 2016-05-14:
clue, it's happening here on line 680 of esp8266_drv.c

        } else {
                /* Client should be inactive.  Disable as needed */
                if (ci->has_ap) {
                        /*
                         * If here, client is active.  Leave AP.
                         * STIEG: TODO: Implement Leave AP call. We are done
                         * for now since code not implemented.
                         */
                        return;
                }

brentpicasso, 2016-05-14:
it seems that a failure of AT+CWJAP? might trigger this cascade of failures.

[ESP8266 Driver] Retrieving Wifi Client Info
[serial (Aux)] TX: AT+CWJAP?(\r)(\n)
[serial (Aux)] RX: (\r)(\n)
[serial (Aux)] RX: ERROR(\r)(\n)
[esp8266] get_ap_cb failed.

brentpicasso, 2016-05-14:
I think I'm working with an ESP12 module with very old firmware. 

[serial (Aux)] TX: AT+GMR(\r)(\n)
[serial (Aux)] RX: 0018000902-AI03(\r)(\n)
[serial (Aux)] RX: (\r)(\n)
[serial (Aux)] RX: OK(\r)(\n)

stieg, 2016-06-02:
I think this was resolved some time ago as I'm not seeing this issue anymore.  Closing.  Re-open if you hit it please.

closed 2016-06-02 by stieg