autosportlabs/RaceCapture-Pro_firmware #618: dropping chars on WiFi RX

brentpicasso, 2016-05-14:
See snippet below, looks like we're dropping characters on RX. May need to implement DMA for this USART. 

-----

[serial (WiFi)] RX: (\r)(\n)                                                    
[serial (WiFi)] RX: SEND OK(\r)(\n)                                             
AT+CIPSEND=0,94(\r)(\n)                                                         
[serial (WiFi)] RX: (\r)(\n)                                                    
[serial (WiFi)] RX: OK(\r)(\n)                                                  
[serial (WiFi)] TX: {"beacon":{"name":"RCT","port":7223,"serial":"FFFF738CFFFFF)
[serial (WiFi)] RX: Recv9 bytes(\r)(\n)                                         
[serial (WiFi)] RX: (\r)(\n)                                                    
[serial (WiFi)] RX: SEND OK(\r)(\n)                                             
AT+CIPSEND=0,94(\r)(\n)                                                         
[serial (WiFi)] RX: (\r)(\n)                                                    
[serial (WiFi)] RX: OK(\r)(\n)                                                  
[serial (WiFi)] TX: {"beacon":{"name":"RCT","port":7223,"serial":"FFFF738CFFFFF)
[serial (WiFi)] RX: Recv 94 bytes(\r)(\n)                                       
[serial (WiFi)] RX: (\r)(\n)                                                    
[serial (WiFi)] RX: SEDOK(\r)(\n)                                               
[at] Command timed out                                                          
[esp8266] send_data_cb failed with msg: Bad response value                      
[ESP8266 Driver] Failed to send data

stieg, 2016-05-25:
Seeing the same behavior screw up my system as well:

```txt
[ESP8266 Driver] Opened comm channel 0
[serial (WiFi)] TX: AT+CIPSTART=0,"UDP","255.255.255.255",7223(\r)(\n)
[serial (WiFi)] RX: 0,CONNECT(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)
[serial (WiFi)] TX: AT+CIPSEND=0,110(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)
[serial (WiFi)] TX: {"beacon":{"name":"RCT","port":7223,"serial":"FFFF847BFFFFFFFF41BEF00F","ip":["192.168.2.229","192.168.4.1"]}}[serial (WiFi)] RX: > (\r)(\n)
[serial (WiFi)] RX: Recv 110 byes(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: SEND OK(\r)(\n)
AT+CIPSEND=0,110(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK> [at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data
[serial (WiFi)] TX: AT+CIPSEND=0,146(\r)(\n)
[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data
```

Notice the "OK>".

stieg, 2016-05-25:
This is worse than I thought.  Seeing lots of data loss through here.  Not sure how much damage its causing but I will be fixing this next since I need to have reliable communications before I can decide what to do with the driver

stieg, 2016-06-02:
And this is Fixed!!!

closed 2016-06-02 by stieg