autosportlabs/RaceCapture-Pro_firmware #630: Investigate sporatic GPS init failures

stieg, 2016-05-18:
@ryandoherty is reporting this on his MK2 unit running 2.10.0 but not 2.9.*.

Here is the log output he got with trace enabled:
```txt
GPS: provisioning attempt
GPS: probing baud rate: 921600
GPS: module detected at: 921600
Unexpected Id: 131
GPS: detected current update rate: 10
GPS: Configuring NMEA messages: Unexpected Id: 168
win
GPS: configure message type: Unexpected Id: 168
win
GPS: configure navigation data message interval: Unexpected Id: 168
win
GPS: Configuring Gnss Navigation Mode: Unexpected Id: 168
win
GPS: Disable NMEA messages: [cell] Unexpected reply. Backing off...
Unexpected Id: 168
win
GPS: provisioned
[cell] Unexpected reply. Backing off...
conn: not connected. retrying
[cell] Resetting modem...
[cell] Unexpected reply. Backing off...
[cell] Unexpected reply. Backing off...
[cell] Unexpected reply. Backing off...
[ESP8266 Driver] Checking Init
[cell] Unexpected reply. Backing off...
[cell] Unexpected reply. Backing off...
conn: not connected. retrying
GPS: timeout
[cell] Resetting modem...
GPS: provisioning attempt
GPS: probing baud rate: 921600
GPS: module detected at: 921600
GPS: detected current update rate: 10
GPS: Configuring NMEA messages: Unexpected Id: 168
Unexpected Id: 168
win
GPS: configure message type: Unexpected Id: 168
win
GPS: configure navigation data message interval: Unexpected Id: 168
win
GPS: Configuring Gnss Navigation Mode: Unexpected Id: 168
win
GPS: Disable NMEA messages: Unexpected Id: 168
win
GPS: provisioned
```

I'll let him comment on the specifics.

ryandoherty, 2016-05-18:
It always happens on 2.10.0 with and without a GPS lock on 2 different mk2s I tested.

One mk2 had the same issue on 2.9.1

stieg, 2016-06-02:
With some of the GPS fixes that just went in with the DMA bits I would be willing to bet these are gone now.

@ryandoherty Want to try a bleeding edge version of 2.10 on a problem unit?

stieg, 2016-06-02:
I have sent you a test build in our firmware channel.  Give it a shot if you have time.  I am going to assume this as provisionally closed since I can't reproduce it on my end, but will leave it open as a reminder to test.

stieg, 2016-06-02:
@ryandoherty confirmed that the patch fixed the issue.  Going to back-port it and the GPS fixes to 2.9.2.  Thanks Ryan

closed 2016-06-02 by stieg