autosportlabs/RaceCapture-Pro_firmware #634: Investigate what it would take to embed all track locations into firmware.

stieg, 2016-05-23:
We want the bare minimum track configs in firmware so that users don't have to load up track info to have things work.  Here is how it would work:

* Upon GPS lock, unit checks to see if a manual track has been given.  If so, use it.
* If no manual track loaded, look at stored tracks to find closest track to location.  If one meets heuristic values, use it.
* Failing the above, look to this embedded DB to find the closest start/finish point to unit location.  If one is reasonably close, load up that data as the track config automatically.

This is in response to issues noted in the field report from Vodden the hell 2016 from @ryandoherty  and myself.  Ideally this DB would be very minimal having only the following:

* Track ID
* Track Start/Finish

Not sure if this is possible; we may also need track Name information etc.  This information would be embedded in the firmware so that our stuff "Just Works" when it gets on a user's unit.  This would solve > 95% of configuration issues most likely since users would have working timing and predictive timing out of the box, paralleling an AIM solo experience.  Being feature level with this unit is critical as its what users expect in the field.

brentpicasso, 2016-05-23:
We have a big chunk of unused flash memory adjacent to the current block holding tracks that could be expanded to increase the number of available tracks.

brentpicasso, 2016-05-23:
On the RaceCapture (RCT) side this is is more support for moving the track database and ideally even lap timing out of the firmware and into the app proper, since RCT requires the app for functionality.

stieg, 2016-05-23:
> On the RaceCapture (RCT) side this is is more support for moving the track database and ideally even lap timing out of the firmware and into the app proper, since RCT requires the app for functionality.

Unfortunately I fail to see what is really gained by doing this since we will still have to support in device track support on the MK2 and MK3 units because of the logging to SD feature and ability to stream telemetry from the device.  To get rid of this properly would require us to have lap code in

* The App
* The Podium Site
* Any supported system that can read our SD card data since the sd card data would no longer have app information.

I guess what I'm trying to say is that to implement your proposed solution is quite a large commitment that requires a fair bit of thought and planning.  We should discuss it at our standup.

brentpicasso, 2016-05-23:
I was referring to RCT specifically on this comment. But yes, can be detailed in the next meeting.

stieg, 2016-06-16:
After investigation and a planning meeting with Ryan and Brent we have a plan of attack for this.  All issues that are part of this plan are as follows: #681 #683 #536.  Because we have a plan I am thus closing this ticket.

closed 2016-06-29 by stieg