autosportlabs/RaceCapture-Pro_firmware #637: build firmware 2.9.1

mach1000, 2016-05-26:
I have a linux dist running to compile the firmware. I have a 2.8.5 which compiles easily but with the new 2.9.1 I get an error which I have pasted in to this.

It is probably a error in my side but advice is very appreciated. 

Thanks
================

 loggerApi_test.cpp:1315:Assertion
Test name: LoggerApiTest::testGetVersion
assertion failed
- Expression: ((string)(String)json["ver"]["git_info"]).size()

loggerApi_test.cpp:1333:Assertion
Test name: LoggerApiTest::testGetStatus
assertion failed
- Expression: ((string)(String)json["status"]["system"]["git_info"]).size()

Failures !!!
Run: 130   Failure total: 2   Failures: 2   Errors: 0
make[3]: *** [test-run] Error 1
make[3]: Leaving directory `/home/mads/ASL/Firmware/asl_2.9.1/RaceCapture-Pro_firmware-master/test'
make[2]: *** [test-run] Error 2
make[2]: Leaving directory `/home/mads/ASL/Firmware/asl_2.9.1/RaceCapture-Pro_firmware-master'
make[1]: *** [test-pristine] Error 2
make[1]: Leaving directory `/home/mads/ASL/Firmware/asl_2.9.1/RaceCapture-Pro_firmware-master'
make: *** [package] Error 2

stieg, 2016-05-26:
What platform are you on?  What version of GCC are you running?  Did you follow exact steps outlined in https://github.com/autosportlabs/RaceCapture-Pro_firmware ?  Show me the output of `g++ --version`.

What this is telling me is that one of our unit tests is failing on your platform specifically, but not one of our official build platforms.

stieg, 2016-05-26:
Also try building on a clean environment.  Run `make clean` followed by `make`

stieg, 2016-06-02:
Haven't heard anything about this @mach1000 any progress?  I'm gonna close this soon otherwise.

mach1000, 2016-06-03:
I am sorry, I haven't had time to try your proposal, but I will do later today.

Thanks

Mvh 

Mads S. Christiansen

Sendt fra min iPhone

mach1000, 2016-06-03:
I am on Ubuntu desktop 14.04.
G++ --version is>
g++ (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4

Just tried but no luck, I will continue debugging sice it could be a mistake of mine.

Thank you

stieg, 2016-06-03:
Interesting.  I just did fix a bug an issue with GCC 4.8.5 and an optimization error.  Wonder if this is related.  See 05e51ac050db9edf1532355c6270e110940d4877.  Try building r/2.10.0 and see if that works for you.  If so it is possibly related.

stieg, 2016-06-05:
Also do re-check the README.  I just did update it with some fixes so perhaps that was the issue.

stieg, 2016-06-05:
Also... can you run `git describe --dirty --always` and paste the output here.  Would be good to know where you are in the build tree.

stieg, 2016-06-09:
Closing due to inactivity.

closed 2016-06-09 by stieg