autosportlabs/RaceCapture-Pro_firmware #638: Finalize WiFi JSON commands

stieg, 2016-05-26:
Need to finalize these so @ryandoherty can add the support to the app properly.  The only major changes I forsee from the way they are now are as follows:

* Will move the JSON key names back to camelcase per @brentpicasso request
* Need to add an enable/disable switch for the entire wifi system (useful for testing etc).
* Probably will consolidate the entire configuration step into a single JSON command.  Currently there are two separate commands.

@ryandoherty if you have feedback on any of this lets chat or leave it here.

stieg, 2016-06-02:
Ok... here is what I expect our JSON for WiFi to look like:
```
{
"wifi":{
	"active": <bool>,
	"client":{
		"active": <bool>,
		"ssid": <String - 24 Chars Max>,
		"passwd": <String - 24 Chars Max>
	},
	"ap":{
		"active":<bool>,
		"ssid": <String - 24 Chars Max>,
		"passwd": <String - 24 Chars Max>,
		"channel":<Number>,
		"encryption":<String - <none|wpa|wpa2|wpa/wpa2>>
	}
}
}
```

ryandoherty, 2016-06-02:
Looks good! Two thoughts:  

* I'd prefer to use 'password' since it's no longer than 'encryption' and improves readability/clarity. 
* The root command does not follow the in use format of 'set[X]Cfg'. We should probably stick with that format for consistency.

brentpicasso, 2016-06-02:
if we abbreviate it we might as well use 'pwd' - nobody should be able to mistake that.

agree on the setXCfg.

stieg, 2016-06-02:
> I'd prefer to use 'password' since it's no longer than 'encryption' and improves readability/clarity.
No problem.  Will change.
> The root command does not follow the in use format of 'set[X]Cfg'. We should probably stick with that format for consistency.
Ok.

With the above changes it would look like this:
```
{
"setWifi":{
	"active": <bool>,
	"client":{
		"active": <bool>,
		"ssid": <String - 24 Chars Max>,
		"password": <String - 24 Chars Max>
	},
	"ap":{
		"active":<bool>,
		"ssid": <String - 24 Chars Max>,
		"password": <String - 24 Chars Max>,
		"channel":<Number>,
		"encryption":<String - <none|wpa|wpa2|wpa/wpa2>>
	}
}
}
```
and
```
{"getWifi":1}
{
"getWifi":{
	"active": <bool>,
	"client":{
		"active": <bool>,
		"ssid": <String - 24 Chars Max>,
		"password": <String - 24 Chars Max>
	},
	"ap":{
		"active":<bool>,
		"ssid": <String - 24 Chars Max>,
		"password": <String - 24 Chars Max>,
		"channel":<Number>,
		"encryption":<String - <none|wpa|wpa2|wpa/wpa2>>
	}
}
}
```

stieg, 2016-06-02:
Blegh to camelcase in JSON.  Not my favorite at all :(

ryandoherty, 2016-06-02:
I'm against abbreviations unless absolutely necessary. I don't think there's a strong technical reason for them, and shortening them can lead to more confusion. kthxbai

stieg, 2016-06-02:
+1 to spelling it out.  PWD == present working directory to me.  I get what you are going for, but not worth it IMHO.  Use `password` and be done

brentpicasso, 2016-06-02:
should be setWifiCfg / getWifiCfg to be consistent with other API commands.

agreed on spelling out password.

stieg, 2016-06-02:
I updated the comment post my initial reply.  It should be up to date now.

ryandoherty, 2016-06-02:
👍

brentpicasso, 2016-06-02:
looks good!

stieg, 2016-06-03:
Merged

closed 2016-06-03 by stieg