autosportlabs/RaceCapture-Pro_firmware #639: Support Measuring Lower RPM Rates

stieg, 2016-05-26:
We need to be able to measure lower RPM rates with accuracy.  As documented by autosportlabs/RaceCapture-Pro_firmware#516 the lowest single pulse RPM rate we support as of 2.9.0 is currently 90 RPM.  I propose we adjust our prescalars so that the following is true:

Speed | Lowest Supported Hz Measurement
-------- | -----------------------------------------------
Fast | 10 Hz
Medium | 1 Hz
Slow | 0.1 Hz

This should give us the breadth of measurement we need to be extremely flexible while only minimally sacrificing precision (which is frankly quite high as is).  This would change the tables to something like this:

Speed | Prescalar | Lowest (Hz) | Lowest (RPM) | Highest (Hz) | Highest (RPM)
-------- | ------------ | --------------- | ----------------- | ---------------- | -------------------------
Fast | 256 |10 | 600 | 656250 | 39375000
Medium | 2562 | 1 | 60 | 65573 | 3934426
Slow | 25200 | 0.1 | 6 | 6666 | 400000

This looks very reasonable since folks can use the fast speed to retain the same high precision RPM measurements they have today.  They can use the medium for slower engines or faster shafts.  And the slow speed is available for anyone measuring something that turns at some very low rate for some unknown reason.

brentpicasso, 2016-05-26:
looks like a good compromise.

stieg, 2016-06-03:
I messed up my original prescalar calculations apparently.  I have since updated them.  But the lowest measurable Hz remains unchanged.

stieg, 2016-06-05:
Merged

closed 2016-06-05 by stieg