autosportlabs/RaceCapture-Pro_firmware #642: SD card write error

fredzyda, 2016-05-28:
[RC_0.LOG.zip](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/287983/RC_0.LOG.zip)
I've had very consistent failures to write to the SD card in firmware version 2.9.0 and 2.9.1. Here is some log output when the issue is occurring:

```
Logging started
fileWriter.c: Start
fileWriter.c: Opening log file
fileWriter.c: FS init success.  Opening file...
fileWriter.c: Opening new log file
fileWriter.c: Opened rc_0.log
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: FS init error: 3
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Msg type 0 failed with code 1
fileWriter.c: Opening log file
fileWriter.c: FS init success.  Opening file...
fileWriter.c: Opening log file rc_0.log
fileWriter.c: Opened rc_0.log
fileWriter.c: flush
(failed)
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
(failed)
(failed)
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: FS init success.  Opening file...
fileWriter.c: Opening log file rc_0.log
fileWriter.c: Opened rc_0.log
fileWriter.c: flush
fileWriter.c: flush
```

I have a SanDisk 32gb SDHC card and the failure seems to happen about once every two seconds. I've attached the log written to the SD card while these errors were printing out and you can see strange things like out of order timestamps and intervals between recording that are sometimes 5x the expected amount.

stieg, 2016-05-28:
Hey Fred,

   Thanks for reporting the issue.  Have you tried reformatting the card?
How did you format it originally?  Are you using FAT 32 FS?  Let me know.
Will you also attach your saved config so we can and reproduce the problem

fredzyda, 2016-05-28:
@stieg, I have not explicitly tried reformatting, but I can do that next. I have verified that the card is in fact FAT 32 and it seems to work just fine in my macbook's built in reader. Here is my config attached. I don't think there's anything super weird going on here, I don't even have any lua scripts written yet.

[firstWorkingMay2016.rcp.zip](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/288050/firstWorkingMay2016.rcp.zip)

stieg, 2016-05-29:
Cool. Thanks for the data dude.  I'll take a peek and let you know.  So just to confirm you are seeing the same output on 2.9.0 **and** 2.9.1?  I ask because there were a few SD card related fixes that went into 2.9.1.

fredzyda, 2016-05-29:
That is correct. I saw the same behavior on both firmware versions.

brentpicasso, 2016-05-29:
Hi Fred,

Does this happen consistently with any of the SD cards you have? Would be interesting to know if it's isolated to that card in your collection.

Thanks for working with us on this.

fredzyda, 2016-05-29:
@brentpicasso, glad to help. I just tried another SD card brand new out of the box and had the same issue. The first time I went to log, it actually hung during the filesystem remount and wasn't able to log again until I power cycled the device. That log looks like it stopped writing mid line in the file. After power cycling the file system remounting message occurred much more frequently, but it still did write something to disk. I've attached both log files off of the SD card. Is there another log I can gather that will give you more information?

A quick note about the setup I was just trying: this is with the car mostly off. I have the main power on but the ignition off, so the only things in the car getting power are the ECU for keeping the volatile memory good, the voice radio for the driver (but I wasn't transmitting), and the RCP. The car was also hooked up to a battery tender, but this should be just about the most electrically friendly situation the device will ever see. It is also mounted a couple of feet away from all the other major electronics in the car and has its own separate ground. The SD cards I tested were both of the same variety. They're 32 GB SanDisk Ultra SDHC cards. Here's the amazon link to the thing I bought:

http://www.amazon.com/SanDisk-microSDHC-Standard-Packaging-SDSQUNC-032G-GN6MA/dp/B010Q57T02

For both of these cards I've used the filesystem they came with, which my mac reports as FAT32. I'm going to try and reformat one to see if that helps.

[Archive.zip](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/288529/Archive.zip)

fredzyda, 2016-05-29:
Just reformatted the second card and got the same behavior:

```
Logging started
fileWriter.c: Start
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
Logging stopped
```

Again this one seemed unable to actually reopen the file after the first few failures.

brentpicasso, 2016-05-29:
Thanks for the update, especially useful that it was a new card out of the box. If you have any others it would be good to try all of them, since SD cards vary quite incredibly in quality and consistency (you should see the testing pile we have).

Also, is there anything elaborate about your RCP setup? Complex Lua script?   If you know how to get to the RCP command line over minicom / realterm / hyperterminal and try doing a configReset (save your config first) to restore to a factory config, and then try writing again. 

let us know, we'll work to get this resolved asap. thanks!

fredzyda, 2016-05-29:
I posted the config file above, you're welcome to examine it for strangeness. This most recent race was the very first I used it in and I was tight on time, so there's nothing fancy going on. There is effectively no lua script at all, just the default onTick() with no contents. I have the 2.5G cellular module configured and running as well as the bluetooth module, but my tablet is currently not connected to it. I'm running the device off of 12V power via the car battery because I know I can't run it with USB alone with the cell modem going. I'm not currently aware of any wiring problems or anything like that, it is all new and reasonably carefully done. It also appears to be gathering data that makes sense.

I'm not currently familiar with connecting to the RCP via a serial terminal, however I remember reading about that in one of your docs, so I'll go find that again and set it up. I don't anticipate having trouble with that. I've saved my config and I'll try the factory reset as soon as I get the serial terminal up and running.

Thanks for all your help! I hope I'm not ruining your weekend!

brentpicasso, 2016-05-29:
it's pretty easy, just ensure you don't have the app running and then connect to the virtual COM port that RCP creates on the system. Hit enter and you'll see a list of options.

Thanks!

fredzyda, 2016-05-29:
That was in fact easy. I just used screen and didn't even have to install anything. With the factory config, the error appears to happen less frequently, but still happens. I also saw a new error output I hadn't seen before. Here's the log output:

```
Logging started
fileWriter.c: Start
fileWriter.c: Opening log file
fileWriter.c: FS init success.  Opening file...
fileWriter.c: Opening new log file
fileWriter.c: Opened rc_2.log
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: FS init error: 3
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Msg type 0 failed with code 1
fileWriter.c: Opening log file
fileWriter.c: FS init success.  Opening file...
fileWriter.c: Opening log file rc_2.log
fileWriter.c: Opened rc_2.log
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: FS init success.  Opening file...
fileWriter.c: Opening log file rc_2.log
fileWriter.c: Opened rc_2.log
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
fileWriter.c: flush
Logging stopped
```

fredzyda, 2016-05-29:
Tried the third SD card I have (same type as the last two, also fresh out of the packaging) and also got errors:
```
Logging started
fileWriter.c: Start
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: FS init error: 3
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
fileWriter.c: Remounting FS due to write error.
fileWriter.c: Opening log file
fileWriter.c: Opened rc_0.log
Logging stopped
```

I never saw the flush message, yet the card had a file with valid looking log data on it when I examined it.

stieg, 2016-06-09:
Assigning this to @brentpicasso and punting on this for now since this may be a hardware issue and I am not able to easily reproduce.  Will re-evaluate after brent diagnoses the unit

stieg, 2016-08-16:
Fixed by commit dc0736c65da4f87803f2a08ee39442c5f4dc8385.  Tested with your SD card @fredzyda .  Works like a freaking charm now.

closed 2016-08-16 by stieg

fredzyda, 2016-08-18:
Nice! Thanks @stieg!