autosportlabs/RaceCapture-Pro_firmware #654: building firmware v2.9.1

psallandre, 2016-06-04:
Hi,
I'm trying to build the firmware, I'm following instructions of Readme for MK2:
I checked out v2.9.1
Installed toolchain: ok
Compile Lua (make PLAT=stm32 generic): ok after having installed 32bit lib: sudo apt-get -y install lib32ncurses5. (I'm on Ubuntu 15.10 64bits)
cd platform/mk2 (instead of stm32_base... readme needs an update)
make PLAT=stm32
`../../src/logger/loggerApi.c:176:38: error: 'MAJOR_REV' undeclared`

So I searched a bit and did from project root:
`make mk2-package`
and indeed I have a firmware in RaceCapturePro_MK2_2.9.1.zip.

But I did a file compare between the official 2.9.1 firmware and the one I just built, and there are lots, really lots of differences and mine is 28 bytes bigger.
So is it normal to have so many differences from one built to another one ? Do you think my built firmware is safe to be flashed?
Of course, my end goal is to flash a custom firmware!
Thanks,
Patrice

jordan81, 2016-06-04:
In the same boat and having the same issue...Don't have the RCP here to test out , but would love to know the answer

stieg, 2016-06-04:
I have now updated the README.  Should be accurate enough now (and much easier to build compared to the way things used to be).  Give that a go now and let me know.

closed 2016-06-04 by stieg

psallandre, 2016-06-04:
Thanks for helping!
So I tried the new toolchain, but I'm afraid there is something else wrong, the binary is still much different from official firmware.
BTW, I tried to build several times, the binary produced is each time exactly the same.
Maybe it's "ASL F4 Loader Project ", as I was still using the one from previous readme.

I tried to follow new readme to build it but I have an error with make :
`Makefile:32: *** BOARD is not defined.  Pass it in as BOARD= or create a config.mk file.  Stop.`
So I tried `make BOARD=mk2`, win!

I installed it with `sudo pip install asl_f4_loader-0.0.7.tar.gz` but now I have an error when building RCP firmware:
```
Creating upgrade images
asl_f4_fw_postprocess -f main.bin -o 0x08020000 -b main.bin -i main.ihex
Usage: asl_f4_fw_postprocess [options]
asl_f4_fw_postprocess: error: No Elf specified
Makefile:125: recipe for target 'main.ihex' failed
```

psallandre, 2016-06-04:
So I went back to previous version of ASL F4 FW PostProcess

I made a change to platform/mk2/Makefile, I add that:
`LD      := $(PREFIX)-ld`
It seems that it was using system wide ld instead of toolchain...
Indeed I have far less differences with official firmware, but still some.

stieg, 2016-06-05:
Ah yes... There is a tagging issue in that project currently.  Need to square away those versions and fix the tagging.  I filed autosportlabs/ASL_F4_bootloader#8 to track that.

psallandre, 2016-06-05:
ok I built ASL_F4_bootloader with the tag you provided, but still no luck, the firmware binary is still different.
I'll send it to you by email, if you can have a look?

jordan81, 2016-06-05:
Thanks for updating the documentation. I had no problems building the firmware

psallandre, 2016-06-05:
@jordan81 good you succeed in building the firmware!
Have you checked the firmware you built is binary strictly equivalent, with say WinMerge or equivalent?
The building process gives some CRC numbers, mines are these:
```
Application CRC: 0x162cc80cL
Info CRC: 0xa272c5b8L
Application length: 270660
```
What are yours?
Could you share you build machine setup, Ubuntu version, etc ?
Thanks

jordan81, 2016-06-05:
I have not yet checked the firmware, but I will later this afternoon and can report back. I’m using a vagrant VM supplied.

stieg, 2016-06-06:
Here is some output for you @psallandre that may help.  This is me building 2.9.1 on my machine.  Note there is v2.9.1 tag and 2.9.1 tag.  Both point to the same change.

```
[stieg@lucious firmware]$ git describe --dirty --always
2.9.1
[stieg@lucious firmware]$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601]
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[stieg@lucious firmware]$ make mk2-package
...
  CC      util/newlib.o
  LD      main.elf
  OBJCOPY main.bin
  Creating upgrade images
asl_f4_fw_postprocess -f main.bin -o 0x08020000 -b main.bin -i main.ihex
Found info block @ offset: 0x31b70
Application CRC: 0x13d9e77L
Info CRC: 0xacf5e005L
Application length: 270644
make[2]: Leaving directory '/home/stieg/Devel/asl/firmware/platform/mk2'
make[1]: Leaving directory '/home/stieg/Devel/asl/firmware'
./bin/package_release.sh MK2 "2.9.1" platform/mk2
  adding: RaceCapturePro_MK2/ (stored 0%)
  adding: RaceCapturePro_MK2/RaceCapturePro_MK2_2.9.1.ihex (deflated 65%)
  adding: RaceCapturePro_MK2/HOW_TO_FLASH.txt (deflated 56%)
  adding: RaceCapturePro_MK2/CHANGELOG (deflated 59%)
[stieg@lucious firmware]$ sha256sum platform/mk2/main.*
da709c3281382d648e8e808dedcbeb078681a75933adb59b74f35463e44a4427  platform/mk2/main.bin
96b4f0bf184c1a2aca49292118a42d0d6701fc2e7ccd40b6d0c2635bf698b297  platform/mk2/main.elf
81d17e4173be7111ef88687c45ed6e2e43980e2d7c3f7352fd08a77ffdf5e948  platform/mk2/main.ihex

```

jordan81, 2016-06-06:
I did flash my 2.9.1 firmware with success, but I did not get a chance to do a diff on it and compare to the official version yet.

psallandre, 2016-06-06:
ok, I have decided to be as brave as Jordan (thanks to you) and try to flash my 2.9.1 firmware.
Success!!
I still don't understand why the firmwares are not binary identical but I won't spend more time investigating that since it's working good.
@stieg, here are the info you requested nonetheless:
```
  CC      util/newlib.o
  LD      main.elf
  OBJCOPY main.bin
  Creating upgrade images
asl_f4_fw_postprocess -f main.bin -o 0x08020000 -b main.bin -i main.ihex
Found info block @ offset: 0x31b70
Application CRC: 0x4835ad11L
Info CRC: 0xe3776810L
Application length: 270652
make[2]: Leaving directory '/home/ps/Documents/Patrice/Git_/RaceCapture-Pro_firmware/platform/mk2'
make[1]: Leaving directory '/home/ps/Documents/Patrice/Git_/RaceCapture-Pro_firmware'
./bin/package_release.sh MK2 "2.9.1" platform/mk2
  adding: RaceCapturePro_MK2/ (stored 0%)
  adding: RaceCapturePro_MK2/RaceCapturePro_MK2_2.9.1.ihex (deflated 65%)
  adding: RaceCapturePro_MK2/CHANGELOG (deflated 59%)
  adding: RaceCapturePro_MK2/HOW_TO_FLASH.txt (deflated 56%)
ps@ps-Ubuntu15:~/Documents/Patrice/Git_/RaceCapture-Pro_firmware$ git describe --dirty
v2.9.1
ps@ps-Ubuntu15:~/Documents/Patrice/Git_/RaceCapture-Pro_firmware$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601]
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ps@ps-Ubuntu15:~/Documents/Patrice/Git_/RaceCapture-Pro_firmware$ sha256sum platform/mk2/main.*
a4956fe176ace6d9e50edb3835f2f6960cbcfb64496eeb871fa8599c0b58ca56  platform/mk2/main.bin
b6fc5880706e0b7b11852def2f35251cc32dd380df1039846fc4443442d86a85  platform/mk2/main.elf
5973c5e070bcc2ab5d3935c2c086e126cba43e28cce21035422654a60b8238e3  platform/mk2/main.ihex
ps@ps-Ubuntu15:~/Documents/Patrice/Git_/RaceCapture-Pro_firmware$ 
```
Thx @stieg for your help!

stieg, 2016-06-06:
That is indeed most strange @psallandre.  My guess is there may be some portion of the build that is using a part of your local GCC toolchain to build the binary instead of using the bins on this toolchain like we should be doing.  I'll use this ticket to track that works.