autosportlabs/RaceCapture-Pro_firmware #657: Lua Scripting Contrib

psallandre, 2016-06-06:
What about creating a repository RaceCapturePro_LuaContrib?
It would be great to have a central repository with some Lua scripts.
Then everyone could contribute with car platform specific scripts or some clever scripts!
Hopefully it would evolve in high value knowledge...

stieg, 2016-06-06:
Not a bad idea honestly.  If you start one with an open license we can always clone the repo into our care and maintain it as well (part of why Git is wonderful).  I have several scripts on my side that would be great to make public.

psallandre, 2016-06-06:
Good you like the idea :)
I don't have mature enough script but when I do, I'll make them public for sure!
In the interval, why not you create this repository and push those great scripts :)? It would make it alive and probably attracts more contributors.

brentpicasso, 2016-06-06:
Great idea!

We created a repo here, feel free to submit Pull requests:

https://github.com/autosportlabs/RaceCapture_Lua_Scripts

Of course, we have our regular, more curated list of Lua scripting examples
here too:

https://autosportlabs.net/RaceCapturePro_Lua_Scripting_Examples

-Brent

psallandre, 2016-06-06:
Cool!
I'll try and feed it as soon as I can

closed 2018-05-17 by brentpicasso