autosportlabs/RaceCapture-Pro_firmware #658: Lua memory limit too conservative + memory monitoring in RaceCaptureApp

psallandre, 2016-06-06:
I have tested a firmware where I have disabled Lua memory ceiling (`LUA_MEM_MAX` set to 0).
A script that was rejected before actually runs great with this firmware.
I have checked memory usage with terminal:
```
- - - Memory Info - - -
Total Memory : 131064
Free Memory : 32336
- - - Lua Info - - -
Lua Top : 7
Lua Memory Usage (KB) : 38
```
I think this memory limit system is too conservative, it is frustrating and so maybe worse than no memory check. What about disabling it until a more accurate memory limit mechanism is found ?

Another related thing, I don't think we can see these memory infos (Free memory and Lua Memory Usage) in RaceCaptureApp? It would be great to have them there. For now I must close RaceCaptureApp, open terminal, connect, type showStats, close terminal, open RaceCaptureApp...

stieg, 2016-06-06:
> I think this memory limit system is too conservative, it is frustrating and so maybe worse than no memory check.

The memory limit exists because without it the system may try to allocate memory in another area, fail, and then crash.  I would rather have it be lower on the safe side than run out and die.  That said we may be able to bump it up by a few K.

I'm curious why do you need so much RAM?  Are you using a lot of persistent variables?  Ideally you want to use as few variables as possible and just use items on the stack as needed.  This keeps the script lean and fast, avoids invocation of the garbage collector, and leads to best possible performance.

> Another related thing, I don't think we can see these memory infos (Free memory and Lua Memory Usage) in RaceCaptureApp? It would be great to have them there. For now I must close RaceCaptureApp, open terminal, connect, type showStats, close terminal, open RaceCaptureApp...

Good feedback.  I filed issue #659 to track that.

kosenko-max, 2016-06-07:
It's a same script that I'm using - Lua can't load script code. It's not
about variables - code itself doesn't fit into memory. Lua code minimizer
and precompiler inside App can potentially solve this problem.

psallandre, 2016-06-07:
yes, indeed, it's the same script as @kosenko-max, so the only persistent variables are channels one. There are about 34 channels.

commit 806e0616 (2017-10-27): increase Lua memory ceiling by 20%. resolves #658
closed 2017-10-27 by brentpicasso