autosportlabs/RaceCapture-Pro_firmware #660: Diagnose and Fix "SEND FAIL" Issue

stieg, 2016-06-07:
After running for about 3 minutes or so I start to notice the following error messages appearing in our log files:
```
[at] Unhandled msg received: SEND FAIL
```
Turning on serial debug we see the following:
```
[serial (WiFi)] TX: AT+CIPSEND=0,104(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)
[serial (WiFi)] TX: {"beacon":{"name":"RCT","port":7223,"serial":"FFFF738CFFFFFFFFB946CE31","ip":["0.0.0.0","192.168.4.1"]}}
[serial (WiFi)] RX: > (\r)(\n)
[serial (WiFi)] RX: Recv 104 bytes(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: SEND OK(\r)(\n)
[serial (WiFi)] RX: SEND FAIL(\r)(\n)[at] Unhandled msg received: SEND FAIL
```
Error is coming from our beacon but nothing is wrong with the beacon itself AFAICT.  Issue may be related to the esp8266 device.  Need to understand the meaning of this message and see how we can make it disappear.  This message shouldn't exist to begin with since this beacon is using UDP with a link-local broadcast address.  So really there is nothing to fail sending wise.  But what is more strange is that the device first sends `SEND OK` followed immediately by `SEND FAIL`.

brentpicasso, 2016-06-07:
Does it get stuck in 'send fail' or does it occur sporadically?

This might help:
http://bbs.espressif.com/viewtopic.php?t=1256

Also, does this also occur with TCP packets?

-Brent

stieg, 2016-06-07:
I saw that thread and it didn't help me much.  This is a UDP beacon, no TCP
involved.  Need to diagnose issue more before I can make any further
statements

stieg, 2016-06-07:
Ahh... I think this may be a regression caused by one of my changes.  Namely just came across the dandy of a Note in the expressif manual: `Note: [<UDP mode>] can only be used when [<UDP local port>] is set`.  I removed those parameters recently, and saw no errors so assumed it was fine.  I would bet that its the cause.

brentpicasso, 2016-06-07:
\o/ ? :)

stieg, 2016-06-07:
> \o/ ? :)

Heh... we will see.  I'm just glad to have someplace to start here :P

closed 2016-06-09 by stieg