autosportlabs/RaceCapture-Pro_firmware #667: Improve API setXXX responses to include error information

ryandoherty, 2016-06-08:
Currently if you attempt to set a configuration and give the RC firmware some invalid data (either a wrong value for a config that exists or some config that doesn't), the firmware returns `-1`. While this tells you that your command failed, you don't know why. Some type of english string that tells you what field(s) were invalid and/or missing, etc would be nice.

Something like: 

`{"status": -1, "field": "ap_encryption", "message": "Invalid encryption value. Valid values are: [none, wpa, wpa2, wpa/wpa2]"}`

stieg, 2016-06-09:
Agreed.  Would be quite nice to have this.  Perhaps a bit of a luxury given the size of the firmware we have to work with, but some basic clues at a minimum would be nice.

ryandoherty, 2016-06-09:
Yeah, I figured this probably isn't 100% feasible, but more than just a `-1` would be nice :)

closed 2018-05-17 by brentpicasso