autosportlabs/RaceCapture-Pro_firmware #679: How to test custom developments ?

psallandre, 2016-06-13:
I have begun creating a new function in src/logger/luaLoggerBinding.c:
`int lua_add_default_channels(lua_State *L)`
that will create some engine default channels and return channels id to Lua.

Is there a way to test it without flashing custom firmware physically? I mean, is it possible to produce an executable instead of a firmware when doing `make mk2`, I could then place a call to that function somewhere in main loop (well I have to exactly where...) to verify it doesn't crash?
More generally, how do you test your developments around Lua scripting ?
Thx

psallandre, 2016-06-14:
I saw there is a rcpsim in test folder, when I launch it:
`RaceCapture/Pro simulator on: /dev/pts/19`
I thought maybe RaceCaptureApp could connect to it, so I installed full dev environment of RaceCaptureApp on my Linux.
RaceCaptureApp is running but it is `Connecting...`
Need some help...

stieg, 2016-06-14:
Hey @psallandre,

> Is there a way to test it without flashing custom firmware physically? I mean, is it possible to produce an executable instead of a firmware when doing make mk2, I could then place a call to that function somewhere in main loop (well I have to find exactly where...) to verify it doesn't crash?

We have a suite of unit tests under the test directory where we do this.  Unfortunately unit tests for Lua doesn't exist at this time.  Issue #214 tracks building a platform where we can register, invoke, and check lua c methods using the lua environment.

> I saw there is a rcpsim in test folder, when I launch it:

Oh man... the sim.  Haven't used that in sometime.  Not sure if it still works but I'm not sure why it wouldn't.

Anyway to achieve the results you would likely need to solve #214 in some fashion.  I know that probably isn't the answer you are looking for.

stieg, 2016-06-14:
Closing since #214 tracks the work that would result from this ticket.

closed 2016-06-14 by stieg

stieg, 2016-06-14:
> Ok...
> About the rcpsim, actually I don't really know if it simulates a RCP as
> the name suggests? Should I be able to connect to it with RC App, if so
> how?

Rcpsim was used to simulate an RCP device so the app could connect to it.
IIRC it creates a /dev/ttyACM* endpoint that the app will search for and
use.  It was supposed to be a way to simulate RCP API behavior on the
machine.