autosportlabs/RaceCapture-Pro_firmware #697: Script fails to load on 2.10.0

brentpicasso, 2016-06-23:
This GoPro script fails to load on 2.10.0
https://wiki.autosportlabs.com/RaceCapturePro2_GoProWiFi

[lua] Loading script. Length: 2647
[lua] Successfully loaded script.
[lua] Script error: [string "--Specify your GoPro wifi password here..."]:30.0: BUG. Please inform AutosportLabs
[lua] Script error: [string "--Specify your GoPro wifi password here..."]:30.0: BUG. Please inform AutosportLabs
[lua] Script error: [string "--Specify your GoPro wifi password here..."]:30.0: BUG. Please inform AutosportLabs
[lua] Failure: Runtime Error

This script works as expected on 2.9.2

stieg, 2016-06-23:
Huh... its choking on the first line of the script yet the parser is printing it out as line 30.  That is odd.

brentpicasso, 2016-06-23:
seems that even a script like this fails:

function onTick()
println("foo")
end

stieg, 2016-06-30:
Heh... the bug seems to be somewhere in the `println` method

closed 2016-06-30 by stieg