autosportlabs/RaceCapture-Pro_firmware #707: If user programs in an empty Lua script, We should reset to default

stieg, 2016-06-28:
Because empty is simply not valid.  Or we could disable Lua entirely if we detect that.

ryandoherty, 2016-06-28:
I think disabling is fine, users might be confused if they configure an empty script, but then see the default onTick() function appear when they read the config.

stieg, 2016-06-28:
Yeah... my thought there as well.

brentpicasso, 2016-06-28:
Vote to simply disable Lua.

Somewhat related, if there's a missing onTick(), we should not consider it an error condition, but just disable the onTick() caller or make it a noop situation. Recently fielded a case where a user was confused by the red error LED.

stieg, 2016-06-28:
> Somewhat related, if there's a missing onTick(), we should not consider it an error condition, but just disable the onTick() caller or make it a noop situation. Recently fielded a case where a user was confused by the red error LED.

@brentpicasso This is what prompted me to file this.  But I don't like your suggestion because if something goes wrong when they enter the script, I want an error condition.  This is why I opted for this behavior.  If there is no script, then there is nothing to do and that is the end of the story.  If there is a script, then behave strictly.  It makes sense and enforces strict LUA behavior so users don't have issues in the field if they have a bad script entered (say a misspelled `onnTick` method).

brentpicasso, 2016-06-28:
I think there are legitimate cases where someone would want a one-time script that didn't include an repetitive task.

If they misspelled onTick() then they would see their repetitive behavior not happening, and diagnose accordingly.

stieg, 2016-06-29:
Plan here per discussion is to simply not error during situation when `onTick` method is not available.  This allows for one time scripts without the need for that callback method and opens the door for a standard of callbacks to be done.

closed 2016-07-04 by stieg