autosportlabs/RaceCapture-Pro_firmware #709: Add capabilities flags to firmware

stieg, 2016-06-29:
As part of our discussion today we will add a "flags" section to the `getCapabilities` JSON method in firmware.  This will return a list of strings that denote the various capabilities of our system, similar to how its done in linux kernel in /proc/cpuinfo

stieg, 2016-07-04:
Ok I know we need the following:

* activeTrack - Denotes ability to set the activeTrack in non-volatile fashion
* wifi - Denotes wifi support
* bluetooth - Denotes Bluetooth support
* telemStream - Denotes api support for telemetry streaming endpoint.

What other flags did we need.  Do you want ones for GPIO and analogue channels etc?  There are some limitations like the fact that we have support for reading the battery, yet don't really have much control over it (other than frequency of updating)

brentpicasso, 2016-07-04:
if you want to be consistent, you can have flags for all hardware capabilities. The other existing capabilities objects can describe how many of some support might exist. 

adc, timer, pwm, imu, gps, racetiming, gpio, can, obd, lua

any other ones?

closed 2016-07-05 by stieg