autosportlabs/RaceCapture-Pro_firmware #734: Honor mph / kph and mi / km units in firmware

brentpicasso, 2016-07-10:
Firmware should honor units value for the following channels:

- distance
- speed

if distance channel's units is 'km' then the telemetry stream / logfile stream should record in km. correspondingly, if 'mi'  then we record in miles.

if speed channel units is 'mph' then we record in miles/hour. correspondingly, if 'kph' then we record in km/hour.

Since we don't have full channel definitions defined in the API for speed and distance channels, we could add members to the GPS config:

'time_ut'
'dist_ut'

these would map to the internal units values for those corresponding channels.

stieg, 2016-07-20:
Looks like we also need to cover altitude to measure between ft and meters.

stieg, 2016-07-20:
Ok.  I intend to use the units outline at https://github.com/autosportlabs/RaceCapture-Pro_firmware/wiki/Units-Supported as a guidline for this.  Here is how it will work:

* Altitude: Valid units will be `ft` and `m` (meters).  All other units default to `ft`
* Distance: Valid units will be `km` and `mi` (miles).  All other units default to `mi`
* Speed: valid units are `mph` and `kph`.  All other units default to `mph`

This gets us two items: It solves the ticket and starts us down the road of strict units.

closed 2016-07-22 by stieg