autosportlabs/RaceCapture-Pro_firmware #753: Diagnose and Fix OSX USB issue with 2.10.0

stieg, 2016-07-19:
For some strange reason the firmware is not getting enumerated on OSX 10.11.5.  Works on 10.11.1 so there is likely some race condition causing this.  2.9.2 works just fine.  Silly macs.

stieg, 2016-07-19:
Bisected firmware tree with @ryandoherty . Issue starts at commit 590d3634fec8941634841e618f0538c63d91c263, yet it is difficult to fathom how that the code change is the root cause (it defines a previously undefined variable).  Further diagnosis shows that the resulting image goes from 305KB to 330KB.  Upon sending @ryandoherty the same build with the debug turned off (set a local annotation point), the build worked on his machine.  So we are obviously hitting some form of assertion or exceeding flash boundaries.

ryandoherty, 2016-07-20:
After much debugging this *seems* to be a problem when RCP is upgraded from 2.9.2 GA => 2.10.0 RC1 with debug *on*. If the upgrade is 2.9.2 => 2.10.0 with debug off, it works.

ryandoherty, 2016-07-20:
Further information: if I do a `resetConfig` on my working 2.10.0 mk2, the problem comes back :(

stieg, 2016-07-21:
Latest updates here are that I am able to reproduce behavior where bootloader on RCT is wiped when builds from Jenkins are flashed.  So need to diagnose the Jenkins issues and see where we can go next.

stieg, 2016-07-27:
So @ryandoherty has mailed me his affected MK2 unit.  Going to pull it apart and diagnose locally.

stieg, 2016-07-29:
Workaround is PR #774, Culprit issue tracked in #775

closed 2016-07-29 by stieg