autosportlabs/RaceCapture-Pro_firmware #765: Remove Link Info from Capabilities API

stieg, 2016-07-25:
Because it has been made redundant by the flags object in the same call and since we have not officially released it we should not do so now.

@ryandoherty I just checked version 1.6.0 of the app and I see that there is no apparent support for reading the links object.  Will you confirm this to ensure there are no regressions in app behavior?

ryandoherty, 2016-07-25:
Correct, 1.6.0 does not use the `links` property, but 1.7.0 does. I'll file a ticket, what flag should be used?

stieg, 2016-07-26:
For replacement of the links there is a flag for each link type supported:

* bt - Bluetooth link
* cell - Cellular link
* usb - Universal Serial Bus link
* Wifi - Wireless Fidelity link

Where the links object before always had them defined, this array will only include the flag if that link is supported.

closed 2016-07-29 by stieg