autosportlabs/RaceCapture-Pro_firmware #795: Cannot reconnect via Wifi

brentpicasso, 2016-08-10:
observed on RCT, may appear on MK2.

STR:

Connect via WiFi (e.g. android)
Close app
restart app
observe not connecting

power cycle device
observe successful re-connect

stieg, 2016-08-10:
Wonderful.  Top of tree 2.10.0?  Or something older?

stieg, 2016-08-10:
Hrmm... wasn't able to reproduce this as described, but did notice that if I execute a `sysReset` call the WiFi does not come back up.  Powercycling the device does indeed fix that.

@brentpicasso Can you confirm if any of your actions would have led to some form of a soft reset of the device.

brentpicasso, 2016-08-10:
Correct, top of tree.  Very easy to repro

brentpicasso, 2016-08-11:
Saw the rest of your comments. 

I flashed a non-debug build and used the wifi settings as provided by default.

stieg, 2016-08-17:
This appears fixed with most recent changed.  Closing.

closed 2016-08-17 by stieg