autosportlabs/RaceCapture-Pro_firmware #799: Writing track config causes hang and wifi connection loss

brentpicasso, 2016-08-11:
When writing track config over wifi connection, the wifi connection will drop.

re-connection fails, and after a period of time, RCT goes into the panic blinker, in a pattern of 5 blinks. 

https://www.youtube.com/watch?v=FkQGyetLOAw

(I think the same panic blinker happens with issue #795 )

stieg, 2016-08-11:
Were you writing the track database or the user defined track?

brentpicasso, 2016-08-11:
User defined track config.

stieg, 2016-08-12:
Got it.  Located a few bugs.  Here is the main one:

```
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: 0
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Rx message timeout
[wifi] Rx message timeout
[ESP8266 Driver] Rx Buffer Overflow Detected
[wifi] Rx message timeout
[ESP8266 Driver] Rx Buffer Overflow Detected
[wifi] Rx message timeout
[ESP8266 Driver] Rx Buffer Overflow Detected
[wifi] Rx message timeout
[wifi] Rx message timeout
[ESP8266 Driver] Rx Buffer Overflow Detected
[wifi] Rx message timeout
[wifi] Rx message timeout
```

Looks like the message we are sending ends up being larger than 256 bytes (Rx queue size).  Going to up this and see if its stable along with other fixes.  Man I need to free up some RAM soon... running tight.

brentpicasso, 2016-08-12:
Cool. What I also noticed was that the firmware panicked some time afterwards, with that flash pattern seen in the yt video above.

stieg, 2016-08-12:
That flash pattern indicates that you ran out of memory.  Its supposed to be 4 flashes but it looks like 5 (4 short, 1 long).

Right now I'm seeing esp8266 related issues on my chip.  Seems to send me back bad data after running for a few seconds.  Here is an example:

```
GPS: detected current update rate: 1
GPS: Configuring update rate to 10: win
GPS: Configuring NMEA messages
GPS: Message was successfully received.
GPS: Successfully configured NMEA
GPS: configure message type: win
GPS: configure navigation data message interval: win
GPS: Configuring Gnss Navigation Mode: win
GPS: Disabling NMEA messages
GPS: Message was successfully received.
GPS: Successfully disabled NMEA messages
GPS: provisioned
[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data

[serial (WiFi)] TX: AT+CIPSEND=0,160(\r)(\n)
[serial (WiFi)] RX: �����������������������������������������������������������������������������������������������������������������������������������������������������������t
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data
����������������������������������������������������������������������������������������������[at] Unhandled msg received: �����������������������������������������������������
��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
�[at] Unhandled msg received: �
���������������[at] Unhandled msg received: ���������������
����������������������������������������������������������������������������������������������[at] Unhandled msg received: �����������������������������������������������������
��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
�[at] Unhandled msg received: �
���������������[at] Unhandled msg received: ���������������
Unknown Command- Press Enter for Help.
RaceCapture >
```

So I restart device and see the following again:

```
GPS: Successfully configured NMEA 
GPS: configure message type: win
GPS: configure navigation data message interval: win
GPS: Configuring Gnss Navigation Mode: win
GPS: Disabling NMEA messages
GPS: Message was successfully received.
GPS: Successfully disabled NMEA messages
GPS: provisioned

[serial (WiFi)] TX: AT+CIPSEND=0,96(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)
[serial (WiFi)] TX: {"beacon":{"name":"RCT","port":7223,"serial":"34002D001257334330313020","ip":["192.168.4.1"]}}(\r)(\n)
[serial (WiFi)] RX: > (\r)(\n)
[serial (WiFi)] RX: Recv 96 bytes(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: SEND OK(\r)(\n)
[serial (WiFi)] TX: AT+CIPSEND=0,96(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)
[serial (WiFi)] TX: {"beacon":{"name":"RCT","port":7223,"serial":"34002D001257334330313020","ip":["192.168.4.1"]}}(\r)(\n)
[serial (WiFi)] RX: > (\r)(\n)
[serial (WiFi)] RX: Recv 96 bytes(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: SEND OK(\r)(\n)
[serial (WiFi)] TX: AT+CIPSEND=0,96(\r)(\n)[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data

[serial (WiFi)] TX: AT+CIPSEND=0,160(\r)(\n)
[serial (WiFi)] RX: �����������������������������������������������������������������������������������������������������������������������������������������������������
```

Going to let this thing sit and bit and come back to it.  Not sure what the hell is going on now.  I am suspicious of how fast the app is writing the configuration data back to the device.  It needs to wait for the ack from the last write message before it starts sending the next message.  Else it will overflow the WiFi receive buffer and then things will go straight to hell.

stieg, 2016-08-16:
Unable to hit this anymore.  Probably fixed by a few of the recent patches I have put in.  Marking as unable to reproduce.  Lets re-open if we can reproduce it again.

closed 2016-08-16 by stieg