autosportlabs/RaceCapture-Pro_firmware #808: tab character in channel name creates corrupt meta data

brentpicasso, 2016-08-13:
Related app issue: https://github.com/autosportlabs/RaceCapture_App/issues/1117

STR:
* create analog channel named ABC; in the channel customization popup add a tab character after the name (press tab). Or, edit configuration file directly to inject tab (ASCII 9) into channel name.

* write configuration in app
* power cycle and re-connect
* observe many errors in meta data reported by the app.  JSON appears corrupted:

{
	"meta": [{
				"nm": "Interval",
				"ut": "ms",
				"min": 0,
				"max": 0,
				"prec": 0,
				"sr": 1
			}, {
				"nm": "Utc",
				"ut": "ms",
				"min": 0,
				"max": 0,
				"prec": 0,
				"sr": 1
			}, {
				"nm": "ABC "
				","
				ut ":"
				ratio ","
				min ":0.0,"
				max ":20.0,"
				prec ":2,"
				sr ":5},{"
				nm ":"
				Battery ","
				ut ":"
				Volts ","
				min ":0.0,"
				max ":20.0,"
				prec ":2,"
				sr ":1},{"
				nm ":"
				AccelX ","
				ut ":"
				G ","
				min ":-3.0,"
				max ":3.0,"
				prec ":2,"
				sr ":25},{"
				nm ":"
				AccelY ","
				ut ":"
				G ","
				min ":-3.0,"
				max ":3.0,"
				prec ":2,"
				sr ":25},{"
				nm ":"
				AccelZ ","
				ut ":"
				G ","
				min ":-3.0,"
				max ":3.0,"
				prec ":2,"
				sr ":25},{"
				nm ":"
				Yaw ","
				ut ":"
				Deg / Sec ","
				min ":-120.0,"
				max ":120.0,"
				prec ":2,"
				sr ":25},{"
				nm ":"
				Pitch ","
				ut ":"
				Deg / Sec ","
				min ":-120.0,"
				max ":120.0,"
				prec ":2,"
				sr ":25},{"
				nm ":"
				Roll ","
				ut ":"
				Deg / Sec ","
				min ":-120.0,"
				max ":120.0,"
				prec ":2,"
				sr ":25},{"
				nm ":"
				Latitude ","
				ut ":"
				Degrees ","
				min ":-180.0,"
				max ":180.0,"
				prec ":6,"
				sr ":10},{"
				nm ":"
				Longitude ","
				ut ":"
				Degrees ","
				min ":-180.0,"
				max ":180.0,"
				prec ":6,"
				sr ":10},{"
				nm ":"
				Speed ","
				ut ":"
				mph ","
				min ":0.0,"
				max ":150.0,"
				prec ":2,"
				sr ":10},{"
				nm ":"
				Distance ","
				ut ":"
				mi ","
				min ":0.0,"
				max ":0.0,"
				prec ":3,"
				sr ":10},{"
				nm ":"
				Altitude ","
				ut ":"
				ft ","
				min ":0.0,"
				max ":4000.0,"
				prec ":1,"
				sr ":10},{"
				nm ":"
				GPSSats ","
				ut ":"
				","
				min ":0,"
				max ":20,"
				prec ":0,"
				sr ":10},{"
				nm ":"
				GPSQual ","
				ut ":"
				","
				min ":0,"
				max ":5,"
				prec ":0,"
				sr ":10},{"
				nm ":"
				GPSDOP ","
				ut ":"
				","
				min ":0.0,"
				max ":20.0,"
				prec ":1,"
				sr ":10},{"
				nm ":"
				LapCount ","
				ut ":"
				","
				min ":0,"
				max ":0,"
				prec ":0,"
				sr ":10},{"
				nm ":"
				LapTime ","
				ut ":"
				Min ","
				min ":0.0,"
				max ":0.0,"
				prec ":4,"
				sr ":10},{"
				nm ":"
				Sector ","
				ut ":"
				","
				min ":0,"
				max ":0,"
				prec ":0,"
				sr ":10},{"
				nm ":"
				SectorTime ","
				ut ":"
				Min ","
				min ":0.0,"
				max ":0.0,"
				prec ":4,"
				sr ":10},{"
				nm ":"
				PredTime ","
				ut ":"
				Min ","
				min ":0.0,"
				max ":0.0,"
				prec ":4,"
				sr ":10},{"
				nm ":"
				ElapsedTime ","
				ut ":"
				Min ","
				min ":0.0,"
				max ":0.0,"
				prec ":4,"
				sr ":10},{"
				nm ":"
				CurrentLap ","
				ut ":"
				","
				min ":0,"
				max ":0,"
				prec ":0,"
				sr ":10}]}

stieg, 2016-10-19:
F'in LOL.  You managed to shift all the characters using a whitespace character.  Nice work.

stieg, 2016-10-19:
Ok... Managed to repro this.  Cause appears to be that there is a special list of characters you need to escape in JSON that we do not escape.

```json
{"meta":[{"nm":"Interval","ut":"ms","min":0,"max":0,"prec":0,"sr":1},{"nm":"Utc","ut":"ms","min":0,"max":0,"prec":0,"sr":1},{"nm":"foo "","ut":"Volts","min":0.0,"max":5.0,"prec":2,"sr":1},{"nm":"Battery","ut":"Volts","min":0.0,"max":20.0,"prec":2,"sr":1},{"nm":"AccelX","ut":"G","min":-3.0,"max":3.0,"prec":2,"sr":25},{"nm":"AccelY","ut":"G","min":-3.0,"max":3.0,"prec":2,"sr":25},{"nm":"AccelZ","ut":"G","min":-3.0,"max":3.0,"prec":2,"sr":25},{"nm":"Yaw","ut":"Deg/Sec","min":-120,"max":120,"prec":0,"sr":25},{"nm":"Pitch","ut":"Deg/Sec","min":-120,"max":120,"prec":0,"sr":25},{"nm":"Roll","ut":"Deg/Sec","min":-120,"max":120,"prec":0,"sr":25},{"nm":"Latitude","ut":"Degrees","min":-180.0,"max":180.0,"prec":6,"sr":10},{"nm":"Longitude","ut":"Degrees","min":-180.0,"max":180.0,"prec":6,"sr":10},{"nm":"Speed","ut":"mph","min":0.0,"max":150.0,"prec":2,"sr":10},{"nm":"Distance","ut":"mi","min":0.0,"max":0.0,"prec":3,"sr":10},{"nm":"Altitude","ut":"ft","min":0.0,"max":4000.0,"prec":1,"sr":10},{"nm":"GPSSats","ut":"","min":0,"max":20,"prec":0,"sr":10},{"nm":"GPSQual","ut":"","min":0,"max":5,"prec":0,"sr":10},{"nm":"GPSDOP","ut":"","min":0.0,"max":20.0,"prec":1,"sr":10},{"nm":"LapCount","ut":"","min":0,"max":0,"prec":0,"sr":10},{"nm":"LapTime","ut":"Min","min":0.0,"max":0.0,"prec":4,"sr":10},{"nm":"Sector","ut":"","min":0,"max":0,"prec":0,"sr":10},{"nm":"SectorTime","ut":"Min","min":0.0,"max":0.0,"prec":4,"sr":10},{"nm":"PredTime","ut":"Min","min":0.0,"max":0.0,"prec":4,"sr":10},{"nm":"ElapsedTime","ut":"Min","min":0.0,"max":0.0,"prec":4,"sr":10},{"nm":"CurrentLap","ut":"","min":0,"max":0,"prec":0,"sr":10}]}
```

Instead it appears an extra `"` character appears, which causes the shift, which then breaks everything.

stieg, 2016-10-19:
Well this is turning out to be a giant can of F'ing worms.  Turns out we never did any JSON escaping at all....  So implementing that now...

closed 2016-10-21 by stieg