autosportlabs/RaceCapture-Pro_firmware #809: Lap timing / predictive timing stops working

brentpicasso, 2016-08-14:
Saw evidence of lap timing stopping completely on this event, despite GPS having a strong signal; GPS position updating and speed. (assuming distance as well, but would need to check the raw logs).

https://podium.live/events/chumpcar-the-24-hour-classic-1/device/307-not-banned-yet

Pattern of behavior: Lap timing / Predictive timing works after a power cycle (e.g. pit stop), then some time afterwards lap timing and predictive timing stops completely.

We have the configuration file for the RCP; it has a somewhat complex Lua script aside from the usual channel mappings.

Thoughts on reproducing:
Create a simulator that can pump raw data right into the GPS tx/rx port, getting as close to real-world emulation as possible.  

Other, less real-world ideas on simulation:

* Get a log from this session and run those through our standard unit tests
* Create a fake GPS HAL device than can receive data updates from an external source (aux serial port, etc)

stieg, 2016-08-30:
Demoting from 2.10.0 release requirements since this is a bug fix and is thus a candidate for 2.10.1.  WIll try to reproduce before 2.10.0 release.

stieg, 2016-10-20:
Dropping milestone since not a blocker for 2.10.1

brentpicasso, 2018-06-14:
This is likely a false alarm due to the task smash and processor reset bus fixed recently in 2.13.5, causing a delay when the predictive timer is running again (after a lap is completed). 

Closing for now, will re-open if necessary.

closed 2018-06-14 by brentpicasso