autosportlabs/RaceCapture-Pro_firmware #814: Track config for devices with no track db have autodetect set to 1

ryandoherty, 2016-08-16:
RaceCapture with firmware 2.10.0 reports back that autodetect is on:

`{"trackCfg":{"rad":0.0001,"autoDetect":1,"track":{"id":0,"type":0,"sf":[0.0,0.0],"sec":[[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0],[0.0,0.0]]}}}`

It should probably be off since it has no track db.

brentpicasso, 2016-08-16:
So, the auto detect flag should still be there even if there is no track
db. This is the indicator that the app should select the track
automatically.

ryandoherty, 2016-08-16:
The bug is that autodetect is set to 1, not that the key exists.

stieg, 2016-08-17:
Correct.  Need to adjust the default so that if the number of DB entries available is 0, then the auto-detect value is 0.

stieg, 2016-08-18:
Merged

closed 2016-08-18 by stieg