autosportlabs/RaceCapture-Pro_firmware #815: Make precision on gyro channels (yaw / pitch / roll ) zero position

brentpicasso, 2016-08-17:
since we're dealing with hundreds of degrees of rotation / sec, having two decimal points of precision seems excessive. Should represent the value in whole numbers.

stieg, 2016-08-17:
Probably not a bad idea.  We should lump the fix for this in with #422 so that we can see the high value Gyro readouts that trade off sensitivity.  Also you are not dealing with "100s of degrees of rotation a second".  If they are rotating that fast they are likely spinning or going through a hairpin turn.  Still I agree that the sensitivity is not necessary.  Bumping to 2.11 since this is not necessary for 2.10.  If there is time left over I will re-evaluate.

stieg, 2016-08-17:
PR #816 fixes this.

stieg, 2016-08-17:
Merged

PR #816 (2016-08-17): change gyro precision to 0 for usability / clarity
resolves #815
closed 2016-08-17 by stieg