autosportlabs/RaceCapture-Pro_firmware #836: Error sending beacon after some time running

brentpicasso, 2016-08-27:
Caught this in the act, need to reproduce:

--------------------------

. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] Command timed out
[esp8266] join_ap_cb failed.
[ESP8266 Driver] Failed to join network
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Retrieving Wifi Client Info
[at] Command timed out
[esp8266] get_ap_cb failed.
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Setting Wifi Client Info
[ESP8266 Driver] Joining network: RaceCaptureDemo2
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
[at] BUG: Received more messages than can handle. Dropping.
*** Start! ***
watchdog: detected watchdog reset!
CAN0 init @ 500000 win
Start OBD2 task
timebase/logging/telemetry sample rate: 50/25/25
IMU: init res=0
[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Initializing Wifi
[at] Unhandled msg received: lllp
GPS: provisioning attempt
GPS: probing baud rate: 921600
[at] Command timed out
[esp8266] Version info:
	AT version:0.50.0.0(Sep 18 2015 20:55:38)
	SDK version:1.4.0
	compile time:Sep 18 2015 21:30:56
[esp8266] Initialized
[ESP8266 Driver] Initialization successful
[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Checking WiFi Device...
[... 146 lines cut ...]
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon
[esp8266] connect_cb failed.
[ESP8266 Driver] Failed to connect: 255.255.255.255
Unable to create Serial for Beacon

stieg, 2016-08-27:
What version of the firmware?

brentpicasso, 2016-08-27:
2.10.0 RC1

stieg, 2016-08-27:
Interesting.  RC2 has the fix for the rude URC handling.  Were you writing
configuration information when this happened?  Any details would be most
welcome.

stieg, 2016-08-30:
So I'm of the opinion that this bug is solved by the fact that we fixed URC handling in the 2.10.0-RC2 release.  If we are able to reproduce the issue there then I would say this warrants deeper investigation.  That and since we are about to fix the join issue (Seen in the log above) I don't think there is much here to diagnose otherwise.  So unless this was tested against RC2 I say I'm going to mark this as unable to reproduce and close.  Thoughts @brentpicasso ?

closed 2016-08-30 by stieg