autosportlabs/RaceCapture-Pro_firmware #837: GPS on RCT gets into re-init loop after a few seconds

stieg, 2016-08-30:
Seems that the GPS code gets into a re-init loop after a short amount of time.  Here is some logging:

```
*** Start! ***
Initializing CAN0 with baud rate 500000
CAN init success!
GPS: Initializing...
IMU: init res=0
Start OBD2 task
timebase/logging/telemetry sample rate: 50/50/50
GPS: provisioning attempt
GPS: probing baud rate: 921600
[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Initializing Wifi
[esp8266] Version info:
        AT version:0.50.0.0(Sep 18 2015 20:55:38)
        SDK version:1.4.0
        compile time:Sep 18 2015 21:30:56
[esp8266] Initialized
[ESP8266 Driver] Initialization successful
[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Increasing baud to 921600
[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Retrieving Wifi Client Info
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Client is inactive.
[ESP8266 Driver] Checking WiFi AP
[ESP8266 Driver] Retrieving Wifi AP Info
[ESP8266 Driver] AP info:
         SSID      : RaceCapture 34330313020
         Channel   : 8
         Encryption: 0
[ESP8266 Driver] Checking WiFi AP
[ESP8266 Driver] Retrieving Wifi IP info
[ESP8266 Driver] Checking Client
[ESP8266 Driver] Client is inactive.
[ESP8266 Driver] Checking WiFi AP
[ESP8266 Driver] AP configured properly
[ESP8266 Driver] Station IP: 192.168.4.1
[ESP8266 Driver] Checking Server
[ESP8266 Driver] Starting server on port: 7223
[ESP8266 Driver] Socket connected on channel 0
[ESP8266 Driver] Setup called on allocated channel 0
[ESP8266 Driver] Closing stale Serial object
[ESP8266 Driver] Connected on comm channel 0
[wifi] New external connection: Wifi Chan 0
GPS: probing baud rate: 115200
GPS: Using baud rate: 115200
GPS: module detected at: 115200
GPS: detected current update rate: 25
GPS: Configuring NMEA messages
GPS: Message was successfully received.
GPS: Successfully configured NMEA
GPS: configure message type: win
GPS: configure navigation data message interval: win
GPS: Configuring Gnss Navigation Mode: win
GPS: Disabling NMEA messages
GPS: Message was successfully received.
GPS: Successfully disabled NMEA messages
GPS: provisioned
track: using fixed configGPS: Too many failures.  Reenum
GPS: Initializing...
GPS: provisioning attempt
GPS: probing baud rate: 921600
GPS: probing baud rate: 115200
GPS: probing baud rate: 9600
GPS: Error: could not detect GPS module on known baud rates
GPS: Too many failures.  Reenum
GPS: Initializing...
GPS: provisioning attempt
GPS: probing baud rate: 921600
GPS: probing baud rate: 115200
GPS: probing baud rate: 9600
GPS: Error: could not detect GPS module on known baud rates
GPS: Too many failures.  Reenum
GPS: Initializing...
GPS: provisioning attempt
GPS: probing baud rate: 921600
GPS: probing baud rate: 115200
GPS: probing baud rate: 9600
GPS: Error: could not detect GPS module on known baud rates
GPS: Too many failures.  Reenum
GPS: Initializing...
GPS: provisioning attempt
GPS: probing baud rate: 921600
```

stieg, 2016-09-09:
Finally managed to catch this in the act on an RCT.  Happened shortly after a reset:

```
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x64(d),0x0d(.),0x0a(.),0xa0(.)
[serial (GPS)] RX: 0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e),0x01(.),0xcd(.),0x70(p),0x14(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x60(`),0x0d(.)
[serial (GPS)] RX: 0x0a(.),0xa0(.),0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e),0x01(.),0xcd(.),0x70(p),0x18(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x6c(l),0x0d(.),0x0a(.),0xa0(.),0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e),0x01(.),0xcd(.),0x70(p),0x1c(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x68(h),0x0d(.),0x0a(.),0xa0(.),0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e),0x01(.),0xcd(.)
[serial (GPS)] RX: 0x70(p),0x20( ),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
# About here is where it went wrong.
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x54(T),0x0d(.),0x0a(.),0xa0(.),0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e)
[serial (GPS)] RX: 0x01(.),0xcd(.),0x70(p),0x23(#),0x00(.),0x00(.),0x00(.),0x18(.),0x38(8),0x24($),0x20( ),0x38(8),0x24($),0x3c(<),0xc4(.),0xfb(.)
[serial (GPS)] RX: 0x3c(<),0xe4(.),0x38(8),0xe0(.),0xfb(.),0xc4(.),0x1c(.),0xdc(.),0xe4(.),0x00(.),0xc0(.),0x03(.),0x00(.),0xe7(.),0x1b(.),0xc0(.)
[serial (GPS)] RX: 0xc3(.),0xdb(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0xdb(.),0x00(.),0xe7(.),0xc0(.),0x03(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0xdb(.)
[serial (GPS)] RX: 0x00(.),0xc0(.),0xc3(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0xd8(.),0x24($),0x00(.),0xe7(.),0xc0(.),0xc0(.),0x24($),0x03(.),0x00(.)
[serial (GPS)] RX: 0x1b(.),0xc0(.),0x03(.),0xf8(.),0x24($),0x00(.),0xe7(.),0x1b(.),0xc0(.),0xe0(.),0x24($),0x00(.),0xe7(.),0x1b(.),0xc0(.),0x18(.)
[serial (GPS)] RX: 0x24($),0x00(.),0xe7(.),0x1b(.),0xc0(.),0x00(.),0x24($),0x00(.),0xc0(.),0x38(8),0x24($),0x03(.),0x00(.),0xe7(.),0x1b(.),0xc0(.)
[serial (GPS)] RX: 0x3f(?),0x24($),0x00(.),0xe7(.),0xc0(.),0x00(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0x00(.),0xc0(.)
[serial (GPS)] RX: 0x00(.),0xe7(.),0xc0(.),0x00(.),0xc0(.),0x03(.),0x00(.),0x1b(.),0xc0(.),0xc3(.),0x00(.),0xc0(.),0xd8(.),0x00(.),0xe7(.),0xc0(.)
[serial (GPS)] RX: 0xc0(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0xc7(.),0x18(.),0x38(8),0x24($),0x20( ),0x38(8),0x24($),0x3c(<),0xc4(.),0xfb(.),0x3c(<)
[serial (GPS)] RX: 0xe4(.),0x38(8),0xe0(.),0xfb(.),0xc4(.),0x1c(.),0xdc(.),0xe4(.),0x00(.),0xe7(.),0xc0(.),0xe0(.),0x00(.),0xe7(.),0xc0(.),0x18(.)
[serial (GPS)] RX: 0x00(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x38(8),0x00(.),0x1b(.),0xc0(.),0xc3(.),0x20( ),0x00(.),0xe7(.),0xc0(.),0x00(.),0xe7(.)
[serial (GPS)] RX: 0x1b(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x00(.)
[serial (GPS)] RX: 0x1b(.),0xc0(.),0x03(.),0x00(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x00(.),0xc0(.),0xc0(.),0x27('),0x00(.),0xe7(.),0xc0(.),0xf8(.)
[serial (GPS)] RX: 0x27('),0x03(.),0x00(.),0xc0(.),0x27('),0x00(.),0xc0(.),0x18(.),0x27('),0x03(.),0x00(.),0xe7(.),0x1b(.),0xc0(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x1b(.),0xc0(.),0x03(.),0x38(8),0x00(.),0x1b(.),0xc0(.),0xc3(.),0x20( ),0x00(.),0xc0(.),0x00(.),0x1b(.),0xc0(.),0x00(.),0x1b(.)
[serial (GPS)] RX: 0xc0(.),0x03(.),0x00(.),0x1b(.),0xc0(.),0x3b(;),0x18(.),0x38(8),0x24($),0x20( ),0x38(8),0x24($),0x3c(<),0xc4(.),0xfb(.),0x3c(<)
[serial (GPS)] RX: 0xe4(.),0x38(8),0xe0(.),0xfb(.),0xc3(.),0xc4(.),0x1c(.),0xdc(.),0xe4(.),0x00(.),0xe7(.),0xc0(.),0x00(.),0xe7(.),0xc0(.),0x00(.)
[... cut ...]
[serial (GPS)] RX: 0x20( ),0xdb(.),0x00(.),0xe7(.),0xe0(.),0x18(.),0xdb(.),0x00(.),0xe7(.),0xe0(.),0x00(.),0x03(.),0x00(.),0xe0(.),0x00(.),0xe0(.)
[serial (GPS)] RX: 0x03(.),0x00(.),0xe0(.),0x00(.),0xe7(.),0xe0(.),0x03(.),0x00(.),0x1b(.),0xe0(.),0x00(.),GPS: probing baud rate: 9600

[serial (GPS)] TX: 0xa0(.),0xa1(.),0x00(.),0x02(.),0x02(.),0x00(.),0x02(.),0x0d(.),0x0a(.),GPS: Error: could not detect GPS module on known baud rates
GPS: Too many failures.  Reenum
GPS: Initializing...
GPS: provisioning attempt
GPS: probing baud rate: 921600
0xa0(.),0xa1(.),0x00(.),0x02(.),0x02(.),0x00(.),0x02(.)
[serial (GPS)] TX: 0x0d(.),0x0a(.),
```

stieg, 2016-09-09:
Break down of the above dump shows that this is the last valid message:

```
0xa0(.),0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e),0x01(.),0xcd(.)
[serial (GPS)] RX: 0x70(p),0x20( ),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.),0x00(.)
[serial (GPS)] RX: 0x00(.),0x00(.),0x00(.),0x00(.),0x54(T),0x0d(.),0x0a(.), # Last good message.  No complete message after this.
0xa0(.),0xa1(.),0x00(.),0x3b(;),0xa8(.),0x00(.),0x00(.),0x05(.),0x65(e)
[serial (GPS)] RX: 0x01(.),0xcd(.),0x70(p),0x23(#),0x00(.),0x00(.),0x00(.),0x18(.),0x38(8),0x24($),0x20( ),0x38(8),0x24($),0x3c(<),0xc4(.),0xfb(.)
[serial (GPS)] RX: 0x3c(<),0xe4(.),0x38(8),0xe0(.),0xfb(.),0xc4(.),0x1c(.),0xdc(.),0xe4(.),0x00(.),0xc0(.),0x03(.),0x00(.),0xe7(.),0x1b(.),0xc0(.)
```

Between the last successful message and when we detect failure is the interesting part.  Converting those characters to ASCII (at least what I can), this is what I get:

```
[serial (GPS)] RX: ..p#....8$ 8$<..
[serial (GPS)] RX: <.8.............
[serial (GPS)] RX: ................
[serial (GPS)] RX: ........$....$..
[serial (GPS)] RX: ....$.....$.....
[serial (GPS)] RX: $.....$..8$.....
[serial (GPS)] RX: ?$..............
[serial (GPS)] RX: ................
[serial (GPS)] RX: .......8$ 8$<..<
[serial (GPS)] RX: .8..............
[serial (GPS)] RX: .....8.... .....
[serial (GPS)] RX: ................
[serial (GPS)] RX: ...........'....
[serial (GPS)] RX: '...'...'.......
[serial (GPS)] RX: ...8.... .......
[serial (GPS)] RX: .....;.8$ 8$<..<
[serial (GPS)] RX: .8..............
[serial (GPS)] RX: ................
[serial (GPS)] RX: ............#...
[serial (GPS)] RX: ................
[serial (GPS)] RX: ............$...
[serial (GPS)] RX: $..$.....$....$.
[serial (GPS)] RX: 8$ 8$<.<.8.'....
[serial (GPS)] RX: ...$..$...<$....
[serial (GPS)] RX: ................
[serial (GPS)] RX: ................
[serial (GPS)] RX: ...............;
[serial (GPS)] RX: ....#...........
[serial (GPS)] RX: ...8$ 8$<.<..8..
[serial (GPS)] RX: ................
[serial (GPS)] RX: ...............
```

So this is obviously not NMEA format.  Upon the tries to re-init and probe the baud it seems that 115200 is still the correct baud as that is the baud where we actually get real data.  But again when I convert it to ASCII it is garbage:

```
[serial (GPS)] RX: ................
[serial (GPS)] RX: ................
[serial (GPS)] RX: ...'............
[serial (GPS)] RX: .......<....$...
[serial (GPS)] RX: .#.8$ 8$<.<.8.'.
[serial (GPS)] RX: ................
[serial (GPS)] RX: ................
[serial (GPS)] RX: ................
[serial (GPS)] RX: .......'........
[serial (GPS)] RX: ................
[serial (GPS)] RX: .......$..$..8$ 
[serial (GPS)] RX: 8$<..<..8.......
[serial (GPS)] RX: ....$.$.....$...
[serial (GPS)] RX: ..$..$...$......
[serial (GPS)] RX: ......
```

Upon testing this again I am eventually able to get it into the same failure state.  Here is the raw output:
[... 85 lines cut ...]
[serial (GPS)] RX: ............8..>
[serial (GPS)] RX: ....&..>.8.....>
[serial (GPS)] RX: ..>.............
[serial (GPS)] RX: .8..............
[serial (GPS)] RX: ......88..8....8
[serial (GPS)] RX: ...8............
[serial (GPS)] RX: .8.......&.. ..>
[serial (GPS)] RX: 8...............
[serial (GPS)] RX: .8..............
[serial (GPS)] RX:  ........8......
[serial (GPS)] RX: .........GPS: probing baud rate: 9600

[serial (GPS)] TX: .........
[serial (GPS)] RX: ................
[serial (GPS)] RX: .....GPS: Error: could not detect GPS module on known baud rates
...........
[serial (GPS)] RX: ................
[serial (GPS)] RX: ................
[serial (GPS)] RX: ...GPS: Too many failures.  Reenum
GPS: Initializing...
```

We can see above that this is slightly different than the failure scenario above (Notice that there is no $ characters and we have lots of 0xff bytes in the second failure scenario where the first has many 0x00 bytes.

My guess is there is some race condition being triggered in the GPS unit that is related to how we setup and get this thing going.

stieg, 2016-09-09:
One other thought... Its possible that this is somehow memory corruption.  I don't recall seeing this issue before on RCP/RCT, but also a lot has changed since then

brentpicasso, 2016-09-09:
One thing I remember is that we had a long-ish delay at the beginning of the driver, before init occurred. We've since reduced that delay.  Maybe something to try? Problem is, it's so hard to repeat.

stieg, 2016-09-12:
@brentpicasso are you aware we have a "restart" command for the wifi chip?  May have to look into that further

stieg, 2016-09-13:
Closing for now with the aforementioned patch.  Will re-open if encountered in the wild again.

closed 2016-09-14 by stieg