autosportlabs/RaceCapture-Pro_firmware #844: Interactive Lua causes system to hang (MK2)

brentpicasso, 2016-09-02:
STR:
open an interactive lua session in minicom
typed one character at lua command line
observe hang

LED flash pattern - 4 flashes, on the 4th flash it stays lit until the cycle repeats

top of 2.10.0 / debug build

stieg, 2016-09-02:
Dug into this with debugger.  A few things to follow up on:

* Not sure if we have much of a buffer for the commands since much of that was changed when I moved the USB command handling over to the new rx_buffer code.
* USB stack is most definitely getting smashed since this is taking the entire lua operation when it runs.  Need to find a way to have the LuaTask run this command so we can keep the USB task size small.
* Possible free of a null pointer, but this may also be side effect of stack smash.

@brentpicasso how much is this blocking you?  Not the easiest thing in the world to fix but will poke at it soonish.  I imagine it's not a blocker but mayhaps I am missing something.

brentpicasso, 2016-09-02:
thanks @stieg  =- Not blocking me right now, I can test by editing and re-running the main script.

closed 2016-09-15 by stieg