autosportlabs/RaceCapture-Pro_firmware #846: Periodic stuttering in wifi transmission

brentpicasso, 2016-09-03:
Some times stuttering in data update is observed with wifi transmission.  This might correlate with AT+CIPSEND taking a longer time to respond due to varying network conditions.

This ticket is to track diagnosing this issue. 

Idea 1: (re)explore the use of AT+CIPSENDBUF to improve performance / throughput.

mumchristmas, 2016-10-15:
Some researh here. I set everything to 25hz. This what I got in the end——only 40296 of the 67733 data points are unique. It's about 32% repeated sampling. So effective rate dropped to 17hz.
[datapoint-161015-ColtMa-raw.xlsx](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/531191/datapoint-161015-ColtMa-raw.xlsx)

And here is the detail view of the repeat sampling in 10s ——1 for repeat, 0 for okay.
[image]

brentpicasso, 2016-10-15:
Thank you for this. 

How do you have your wifi configured? Are you connected directly to RC's access point, or is RC connected to the laptop or phone's hotspot, or is RC and client connected through a common router / access point (like your home access point)?

mumchristmas, 2016-10-15:
The client connected directly to RC's access point. My dashcam has a wifi too, so I try to change the channel of RCT wifi to avoid the interference. The result is just same.

brentpicasso, 2016-10-15:
Got it.  when you have time, could you characterize it further under different WiFi environments and conditions? That would help understand where and when it's happening. Also, were you also seeing visual stuttering / hanging? We've noticed this primarily when the WiFi environment is more complicated, such as going through a common access point, or when the wifi network sees congestion. 

Thanks!

mumchristmas, 2016-10-15:
Okay, I'll test it tomorrow. I also met with app hanging & automatic reset in highway once (and only once). Then the App asked me to select the track again.

brentpicasso, 2016-10-15:
Thank you for your contribution!

We are working on a dramatic improvement in how tracks are created and set up, coming in app 1.8.0.

mumchristmas, 2016-10-17:
Update: Client Mode(RCT-->Android) is much stable than AP Host Mode.
In my test, the repeat rate drop to 21% when RCT connected to the Android phone hotspot. Will test with the "RCT--> iOS hotspot <--Android" later.

CullenShane, 2017-02-14:
So I'm not sure what the issue is, but, I have a new datapoint.

Using the RaceCapture App, I charted the number of ms (off of UTC) between each sample.  This is with no OBD or CAN data, only 50Hz GPS.

[image]

Now, here's the same chart (granted, only 700 some samples instead of 4000) generated from SoloStorm, What makes me think this is the same problem is because 60 ms is the period of 17Hz.  Seems like there's a bottle neck somewhere in the RaceCapture while running in AP mode.

[image]

CullenShane, 2017-02-14:
I noticed this discrepancy in GPS time vs System time from SoloStorm, it looks like something confuses the Wifi and it drops out for a second.  Meanwhile the RaceCapture is sampling, (though it's falling behind, which is interesting) and eventually everything catches up and things resume.

I am starting to think that this is a performance issue or bug on the RCT, and the stuttering is a side effect of that.

[image]

brentpicasso, 2017-02-14:
Hi, @CullenShane  good analysis, Thank you. We'll be digging into this, back-tracing where lag might be showing up. Curious how this regression showed up! Should have some answers soon.

mmarich, 2017-02-15:
I'm going to also double-check the amount of time it is taking to process each sample in SS, however since it can stream from an RCP Mk2 via Bluetooth I don't think there should be an issue.