autosportlabs/RaceCapture-Pro_firmware #847: Firmware crash with regular use

brentpicasso, 2016-09-05:
top of 2.10.0, debug build. 

Firmware crashes with a 2 blink panic with regular use over wifi (RCT).  Things that will cause the crash: Connect, operate for a while, disconnect, re-connect, etc will trigger the crash. Will repro with single client

stieg, 2016-09-05:
@brentpicasso Please try and use the output of `git describe` instead of saying "top of 2.10.0".  While I understand your meaning, it is quite imprecise and often doesn't tell me enough.  In this case the panic flashes changed a bit recently, so I need to know if you are using the old style or the new style.

brentpicasso, 2016-09-05:
:+1: 

git describe reports:
2.10.0_RC2-28-g9342906

stieg, 2016-09-05:
Ok.  Here is how you begin to diagnose these issues:

```
enum panic_cause {
        PANIC_CAUSE_UNKNOWN = 1,
        PANIC_CAUSE_ASSERT,
        PANIC_CAUSE_OVERFLOW,
        PANIC_CAUSE_MALLOC,
        PANIC_CAUSE_UNREACHABLE,
        PANIC_CAUSE_TASK_CREATE,
        PANIC_CAUSE_HARDWARE,
        PANIC_CAUSE_SCHEDULER,
};
```

This is in panic.h and is what defines the meaning of the flashes.  In this case it would seem you hit some form of assert.  My guess is that this is a hard fault since that is what the hardfault handler reports back upon panic.

brentpicasso, 2016-09-05:
There's no specific pattern where this reproduces; it does happen fairly quickly though with just wifi connection, disconnects, re-connects, etc.

stieg, 2016-09-06:
Finally got it to reproduce:

```
#0  0x0801c448 in cpu_device_spin (ms=1397, ms@entry=5000) at hal/cpu_stm32/cpu_device_stm32.c:95
#1  0x0800e5f8 in panic (cause=cause@entry=PANIC_CAUSE_MALLOC) at ../../src/util/panic.c:47
#2  0x0800e650 in vApplicationMallocFailedHook () at ../../src/util/panic.c:81
#3  0x0800ee66 in pvPortMalloc (xWantedSize=<optimized out>, xWantedSize@entry=513) at FreeRTOS/Source/portable/MemMang/heap_4.c:244
#4  0x0800f08e in xQueueGenericCreate (uxQueueLength=uxQueueLength@entry=512, uxItemSize=uxItemSize@entry=1, ucQueueType=ucQueueType@entry=0 '\000') at FreeRTOS/Source/queue.c:311
#5  0x0800cbbc in serial_create (name=name@entry=0x802b22a "Wifi Chan 2", tx_cap=tx_cap@entry=512, rx_cap=rx_cap@entry=768, cfg_cb=cfg_cb@entry=0, cfg_cb_arg=cfg_cb_arg@entry=0x0, post_tx_cb=post_t\
x_cb@entry=0x8005d99 <_tx_char_cb>, post_tx_cb_arg=post_tx_cb_arg@entry=0x100004c4) at ../../src/serial/serial.c:134
#6  0x08005ea4 in channel_setup (index=index@entry=2, rx_size=rx_size@entry=0, tx_size=tx_size@entry=0) at ../../src/drivers/esp8266_drv.c:326
#7  0x08006072 in socket_connect_handler (chan_id=2) at ../../src/drivers/esp8266_drv.c:365
#8  socket_state_changed_cb (chan_id=2, action=<optimized out>) at ../../src/drivers/esp8266_drv.c:413
#9  0x08005562 in sparse_urc_cb (msg=0x20006de0 "2") at ../../src/devices/esp8266.c:195
#10 sparse_urc_cb (msg=0x20006de0 "2") at ../../src/devices/esp8266.c:165
#11 0x0800bfea in process_cmd_or_urc_msg (ati=0x100000b4, msg=0x20006de0 "2") at ../../src/modem/at.c:227
#12 0x0800c0a4 in at_task_run_bytes_read (msg=<optimized out>, ati=0x100000b4) at ../../src/modem/at.c:280
#13 at_task (ati=0x100000b4, ms_delay=<optimized out>) at ../../src/modem/at.c:349
#14 0x080056c2 in esp8266_do_loop (timeout=timeout@entry=3) at ../../src/devices/esp8266.c:247
#15 0x0800684e in task_loop () at ../../src/drivers/esp8266_drv.c:1188
#16 task (params=<optimized out>) at ../../src/drivers/esp8266_drv.c:1219
#17 0x0800ebd8 in vPortEnterCritical () at FreeRTOS/Source/portable/GCC/ARM_CM3/port.c:367
#18 0xa5a5a5a4 in ?? ()
#19 0xa5a5a5a4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Really hard to get this to happen.  Occuring because we are trying to open a new connection and have no more resources for the Serial ports.

stieg, 2016-09-06:
Hrmm... I'm also seeing a hardfault happen here.  **Sigh...** its likely something to do with the destruction and reallocation of Serial device resources since it always seems to happen around that time.

brentpicasso, 2016-09-07:
Just saw a flash code 4 testing the latest as of today (2.10.0_RC2-30-g154c12d)

Repro was repeated connects over wifi, disconnect / reconnect type of operations.

stieg, 2016-09-08:
Patch merged.  Seems to have fixed it.

closed 2016-09-08 by stieg