autosportlabs/RaceCapture-Pro_firmware #848: various config writing fail via wifi

brentpicasso, 2016-09-05:
When connected over WiFi, various configs fail to write. Example:

Writing WiFi config
writing OBD2 config

problem does not repro over usb. Maybe stack smash?

stieg, 2016-09-05:
Well this is hard to trace down since I don't know if the device is failing or is returning a failed to accept error code (The app does ignore these replies).  If it was a stack smash you would see a panic since the debug builds check the stack sanity every context switch.

brentpicasso, 2016-09-05:
git describe :  2.10.0_RC2-28-g9342906 

Previously I had also observed it when also writing OBDII config, but now only writing WiFi config makes the problem happen consistently.
No response is observed when writing WiFi config or OBDII config. 

 app is launched as:   ./main.py -d -- --conn_type=wifi

STR: 
connect app via wifi
read config
modify wifi config
write wifi config
observe app displaying timeout message

example output from command line
[DEBUG  ] [SocketComm  ] writing message {"setWifiCfg":{"active":true,"ap":{"active":true,"encryption":"none","password":"","ssid":"RaceCapture 35239383620","channel":11},"client":{"active":true,"password":"xxxxxxxxx","ssid":"spam333"}}}
[DEBUG  ] [RCPAPI      ] Tx: {"getStatus":null}
[DEBUG  ] [SocketComm  ] writing message {"getStatus":null}
[ERROR  ] [SocketConnection] timeout
[WARNING] [RCPAPI      ] Read message timeout waiting for setWifiCfg
[WARNING] [RCPAPI      ] POKE
[DEBUG  ] [SocketComm  ] writing message

stieg, 2016-09-05:
Oh... I wonder if this is causing the WiFi connection to break the connection because its adjusting parameters.  Do you remember what you changed exactly?  For example did you change the client settings?  This would likely cause the WiFi to become unresponsive for a few seconds while it would update the client settings (Happens right after the config is updated).

stieg, 2016-09-05:
Also... thank you for the `git describe` output.  That helps a lot :smile:

brentpicasso, 2016-09-05:
:+1: 

I have RCT connected as a client to home wifi.  I went and disabled the AP (something unrelated), wrote config and verified the issue occurred.

stieg, 2016-09-05:
Ok.  Just played with this a bit.  When I disabled the WiFi Client (I was using the WiFi AP) I saw no issues.  However when I re-enabled it and wrote the config I got a disconnect.  Analysis shows that this is the WiFi trying to connect to the client network, which incurs a multi-second delay in which no data can be sent to the WiFi unit (or at least is not coming through).  Unit remained operational and available.  App was unable to re-connect, which may be related to https://github.com/autosportlabs/RaceCapture_App/issues/1196 .  The hard thing here is I'm trying to determine what is an actual bug vs just expected behavior (user is adjusting wifi settings while using Wifi to configure device).  This seems like it may be expected.  I suppose we should probably ensure that we return a status before altering the WiFi around.  Perhaps that is the fix here.

brentpicasso, 2016-09-05:
Agreed, the problem would be exacerbated by https://github.com/autosportlabs/RaceCapture_App/issues/1196. 

If the user is changing something unrelated to their current connection - e.g. modifying AP settings while they're connected via client, or vice-versa, a good experience would be to not disrupt the current connection.

If we're re-initializing the whole stack (which is a great way to ensure consistent behavior) then possibly sending the response first could be a way to make the experience better.

stieg, 2016-09-05:
Ok was able to reproduce with debug on.  Here is what I saw:

```
[serial (WiFi)] RX: SEND OK(\r)(\n)[ESP8266 Driver] Checking WiFi Device...
[ESP8266 Driver] Checking Client   
[ESP8266 Driver] Setting Wifi Client Info
[ESP8266 Driver] Joining network: LAN Downunder

[serial (WiFi)] TX: AT+CWJAP_DEF="LAN Downunder","<REDACTED>"(\r)(\n)
[serial (WiFi)] RX: 1,CONNECT FAIL(\r)(\n)
[serial (WiFi)] RX: WIFI DISCONNECT(\r)(\n)[ESP8266 Driver] Client disconnected

[serial (WiFi)] RX: WIFI CONNECTED(\r)(\n)[ESP8266 Driver] Client connected

[serial (WiFi)] RX: WIFI GOT IP(\r)(\n)[ESP8266 Driver] Client got IPv4 address

[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)[ESP8266 Driver] Successfully joined network
[ESP8266 Driver] Checking Client   
[ESP8266 Driver] Retrieving Wifi Client Info

[serial (WiFi)] TX: AT+CWJAP_DEF?(\r)(\n)
[serial (WiFi)] RX: +CWJAP_DEF:"LAN Downunder","e0:3f:49:9a:37:e8",6,-41(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)[ESP8266 Driver] Checking Client
[ESP8266 Driver] Retrieving Wifi IP info

[serial (WiFi)] TX: AT+CIFSR(\r)(\n)
[serial (WiFi)] RX: +CIFSR:APIP,"192.168.4.1"(\r)(\n)
[serial (WiFi)] RX: +CIFSR:APMAC,"5e:cf:7f:07:c6:cf"(\r)(\n)
[serial (WiFi)] RX: +CIFSR:STAIP,"192.168.1.252"(\r)(\n)
[serial (WiFi)] RX: +CIFSR:STAMAC,"5c:cf:7f:07:c6:cf"(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)[ESP8266 Driver] Checking Client
[ESP8266 Driver] Client configured correctly
[ESP8266 Driver] Client IP: 192.168.1.252
[ESP8266 Driver] Checking WiFi AP  
[ESP8266 Driver] Updating Wifi AP Config.

[serial (WiFi)] TX: AT+CWSAP_DEF="RaceCapture 34330313020","",8,0(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)[ESP8266 Driver] Checking WiFi AP
[ESP8266 Driver] Retrieving Wifi AP Info

[serial (WiFi)] TX: AT+CWSAP_DEF?(\r)(\n)
[serial (WiFi)] RX: +CWSAP_DEF:"RaceCapture 34330313020","",8,0,4,0(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: OK(\r)(\n)[ESP8266 Driver] AP info:
         SSID      : RaceCapture 34330313020
         Channel   : 8
         Encryption: 0
[ESP8266 Driver] Checking WiFi AP  
[ESP8266 Driver] AP configured properly
[ESP8266 Driver] Station IP: 192.168.4.1

[serial (WiFi)] TX: AT+CIPSEND=1,256(\r)(\n)
[serial (WiFi)] RX: link is not valid(\r)(\n)
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: ERROR(\r)(\n)[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1

[serial (WiFi)] TX: AT+CIPCLOSE=1(\r)(\n)
[serial (WiFi)] RX: UNLINK(\r)(\n) 
[serial (WiFi)] RX: (\r)(\n)
[serial (WiFi)] RX: ERROR(\r)(\n)[esp8266] close_cb failed.
```

When I updated the config I changed the AP settings but did not change the client settings.  I was connected using the client.  As we can see from the logs the client decided to re-configure itself anyways.  This then caused the socket to fail and thus the connection was lost.  That is what can be fixed here

brentpicasso, 2016-09-05:
The other thing to consider is restoring a config from a saved file, which
will affect possibly all settings, and should complete without fail.

stieg, 2016-09-05:
Ah... now I remember why this is not a bug.  This is not really a bug because the user (In this case the app) is specifying new wifi client settings even when nothing has change.  Here is my example:

1.  Start the app in wifi connection mode with debug enabled.
2.  Connect to your unit using Wifi.
3.  When connected change a setting on the AP side of the wifi.  Do not change anything on the client side.
4.  When you click write you will see the following in the debug output:
```
 Execute Sequence begin
[DEBUG             ] [RCPAPI      ] Tx: {"setWifiCfg":{"active":true,"ap":{"active":true,"encryption":"none","password":"","ssid":"RaceCapture 34330313020","channel":8},"client":{"active":true,"password":"<REDACTED>","ssid":"LAN Downunder"}}}
```

See how we are setting _both_ the client and ap settings there?  That is what is causing the issue.  When you specify the client settings we don't necessarily know all of the state on the wifi client (like the wireless password).  As a result every time this gets updated we need to re-send it to the client.  This causes the network to disconnect and the connections to drop.  Instead the app should send the following:

```
```
 Execute Sequence begin
[DEBUG             ] [RCPAPI      ] Tx: {"setWifiCfg":{"active":true,"ap":{"active":true,"encryption":"none","password":"","ssid":"RaceCapture 34330313020","channel":8}}}
```

Notice that there is no client key or object.  This is perfectly acceptable.  It tells our WiFi unit not to update any client settings and leave things the way they are.

The best I can do here is change some code ordering so that we return a response code before issuing the configuration update commands.  Beyond that I'm afraid I can't do much else as this is expected behavior.

stieg, 2016-09-05:
PR up: https://github.com/autosportlabs/RaceCapture-Pro_firmware/pull/850

stieg, 2016-09-05:
Filed sister bug here: https://github.com/autosportlabs/RaceCapture_App/issues/1197 .  One other thought does come to mind.  I am able in the settings are to check if any settings changed, and if not then avoid calling the callback that invokes wireless re-configuration.  This would solve the problem but is probably less than ideal.  I may implement it anyways as its probably faster to solve the issue this way than the other way.

stieg, 2016-09-10:
Merged.

closed 2016-09-10 by stieg