autosportlabs/RaceCapture-Pro_firmware #861: WiFi UDP beacon stops transmitting

ryandoherty, 2016-09-16:
Not 100% reproducible, but seems to occur under these conditions:

- Mk2 (tested on this only so far) 2.10.0-RC3-8-gd7b2f41
- AP mode for WiFi module
- Kindle fire connected to AP
- 1.7.5 app

- Connect to AP from tablet
- Open app, change connection type to wifi
- Watch app connect
- Go to dashboard mode
- Wait a few minutes. After ~5 minutes, the app will see a disconnect, then it will reconnect
- Kill app, clear app preferences/cache via Android/OS settings. Do not uninstall.
- Reopen app. 
- Watch app wait for RCP, even though it is connected to RCP's wifi
- Watch wifi activity LED, it will not blink at all.

ryandoherty, 2016-09-16:
RCP's wifi is still working, if you don't clear the app preferences, it can reconnect because it remembers the IP address previously used to connect to RCP. The UDP beacon is not sending afaict. 

I did a Wireshark sniff from my laptop when connected to RC's AP, no UDP beacons were being sent.

ryandoherty, 2016-09-16:
Seems like the 'trick' to reproducing this is for there to be a disconnect between RCP and the app. Happens randomly and sometimes never. Not sure what causes it or how to repro.

ryandoherty, 2016-09-16:
RCP logs (via USB) from around the time UDP beacons stopped:

```lly received.
GPS: Configuring was successful.
GPS: detected current update rate: 1
GPS: Configuring update rate to 10: win
GPS: Configuring NMEA messages
GPS: Message was successfully received.
GPS: Successfully configured NMEA
GPS: configure message type: win
GPS: configure navigation data message interval: win
GPS: Configuring Gnss Navigation Mode: win
GPS: Disabling NMEA messages
GPS: Message was successfully received.
GPS: Successfully disabled NMEA messages
GPS: provisioned
[USB] Starting telem stream on USB
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Telemetry stream rate too high.  Reducing to 10
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Telemetry stream rate too high.  Reducing to 10
[USB] Stopping telem stream on USB
[USB] Starting telem stream on USB
[at] Unhandled msg received: 1
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[esp8266] close_cb failed.
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Telemetry stream rate too high.  Reducing to 10
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Telemetry stream rate too high.  Reducing to 10
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[esp8266] close_cb failed.
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Telemetry stream rate too high.  Reducing to 10
[wifi] Stopping telem stream on Wifi Chan 1
[wifi] Starting telem stream on Wifi Chan 1
[wifi] Telemetry stream rate too high.  Reducing to 10
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[esp8266] close_cb failed.
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[ESP8266 Driver] Socket closed on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[ESP8266 Driver] Socket closed on channel 1
[... 153 lines cut ...]
[ESP8266 Driver] Channel is not open.  Dropping data
[ESP8266 Driver] Socket closed on channel 1
[at] Command timed out
[esp8266] close_cb failed.
[ESP8266 Driver] Socket connected on channel 1
[ESP8266 Driver] Socket closed on channel 1
[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 0
[wifi] New external connection: Wifi Chan 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[at] Command timed out
[esp8266] close_cb failed.
[ESP8266 Driver] Socket closed on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
```

stieg, 2016-09-17:
This is the important log line here:
```
[ESP8266 Driver] Failed to send data on channel 0
```
Something is killing the socket and internally we are not handling it it would seem.  This is very strange though since UDP is stateless the only thing that should kill a connection is on the local wireless device.  Based on those logs though something is really messing up with your WiFi chip.  Wish I knew what it was.  Will be interested to see if this happens on your RCT.  Can you try the following:

* Open a terminal with a long backlog.  Or one that can somehow record to file.
* Open minicom
* type `setSerialLog 4 1`
* type `flashConfig`
* Power cycle the unit
* When it come back go into minicom ASAP
* type `viewLog`

This should get you a live stream of the log over USB without interfering with the WiFi.  Try to reproduce the problem this way.  If you can you should be able to capture all the messages that lead up to the failure.  That should be sufficient enough for me to diagnose and resolve the issue if there is something I can do to solve it.

ryandoherty, 2016-09-17:
Reproduced it with logs! \o/
[outoput.txt.zip](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/477962/outoput.txt.zip)

stieg, 2016-09-17:
Unfortunately you missed the serial logging step so I can't see what
failed.  Looks like the WiFi unit fails in some odd way.

stieg, 2016-09-18:
Wow... so my previous comment was wrong... I was looking at the wrong output.  Digging...

stieg, 2016-09-18:
Right... the end of the file has the evidence...

```
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,448(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"35004B001851343033393930","git_info":"2.10.0-RC3-8-gd7b2f41","release_type":"development","uptime":1646030},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"","sig_str":0,"number":"","state":"unknown"},"telemetry":{"status":0,"state":"idle","dur":0},"bt":{"init":1},"logging":{"status":0,"dur":0},"track"
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 448 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,314(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: :{"status":0,"valid":false,"trackId":0,"inLap":0,"armed":0},"imu":{"init":true},"wifi":{"active":true,"initialized":true,"ap":{"active":true},"client":{"active":false,"connected":false}}}}(\r)(\n)
[serial (Aux/Wifi)] TX: {"s":{"t":1646000,"d":[1646058,0,4.72,0.03,0.0,0.93,-1,-1,0,0.0,0.0,0.0,0.0,0.0,0,0,0.0,0,0.0,-1,0.0,0.0,0.0,0,16777215]}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[serial (Aux/Wifi)] TX: AT+CIPCLOSE=1(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[ESP8266 Driver] Channel is not open.  Dropping data
(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[ESP8266 Driver] Channel is not open.  Dropping data
(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[ESP8266 Driver] Channel is not open.  Dropping data
1,CLOSED(\r)(\n)[ESP8266 Driver] Socket closed on channel 1
[at] Command timed out
[esp8266] close_cb failed.
[serial (Aux/Wifi)] TX: AT+CIPSEND=0,100(\r)(\n)[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
```

It appears that the WiFi unit gets stuck sending and fails to break out of that command.  The command gets hit with several URCs and then eventually times out and closes the socket internally and tries to close the socket externally but fails to do so.  Not sure how/what is causing this but on the surface it appears to be related to the TCP command taking too long and not timing out properly or something in the Esp8266 chip failing hard.

stieg, 2016-09-18:
So ran my MK2 unit over Wifi all last night and was unable to reproduce this.  Going to grab a fire tablet next and see if that is somehow related.

stieg, 2016-09-19:
Finally managed to reproduce behavior locally.  Here is the logs:

```
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,95(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"s":{"t":71324100,"d":[71324159,0,0.0,0.0,0.0,0.0,0.0,0,0,0.0,0,0.0,-1,0.0,0.0,0,14679555]}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 95 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,124(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"s":{"t":71324200,"d":[71324259,0,-0.03,-0.0,0.97,-0,-0,-1,0.0,0.0,0.0,0.0,0.0,0,0,0.0,0,0.0,-1,0.0,0.0,0.0,0,16777211]}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 124 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,95(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"s":{"t":71324300,"d":[71324359,0,0.0,0.0,0.0,0.0,0.0,0,0,0.0,0,0.0,-1,0.0,0.0,0,14679555]}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 95 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] RX: 1,CONNECT FAIL(\r)(\n)[at] Unhandled msg received: 1

[serial (Aux/Wifi)] TX: AT+CIPSEND=1,124(\r)(\n)
[serial (Aux/Wifi)] RX: link is not valid(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: ERROR(\r)(\n)[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1

[serial (Aux/Wifi)] TX: AT+CIPCLOSE=1(\r)(\n)
[serial (Aux/Wifi)] RX: UNLINK(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: ERROR(\r)(\n)[esp8266] close_cb failed.

RaceCapture/Pro MK2 > viewLog
Starting logging mode.  Hit "q" to exit
```

Looks like an unhandled AT message there then the state gets messed up.  Need to figure out some corner cases here.

stieg, 2016-09-20:
RCA is that this is caused by AT timeouts on sending UDP packets, beacon channel is closed and never reopened.  Need to extend AT command timeout window and add check to reopen beacon channel in event that it is closed.

stieg, 2016-09-20:
Here is the issue clearly seen in the logs (after ANSI strings were stripped out):

```
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,124(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"s":{"t":220000,"d":[220058,0,4.74,0.04,-0.01,0.93,-1,-1,0,0.0,0.0,0.0,0.0,0.0,0,0,0.0,0,0.0,-1,0.0,0.0,0.0,0,16777215]}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 124 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] TX: AT+CIPSEND=0,100(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"beacon":{"name":"RCP_MK2","port":7223,"serial":"35004B001851343033393930","ip":["192.168.4.1"]}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 100 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: SEND OK(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,428(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"status":{"system":{"model":"RaceCapture/ProMK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"35004B001851343033393930","git_info":"2.10.0-RC3-17-g8283fa5","release_type":"development","uptime":220822},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"","sig_str":0,"number":"","state":"unknown"},"telemetry":{"status":0,"state":"idle","dur":0},"bt":{"init":2},"logging":{"status
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[serial (Aux/Wifi)] TX: AT+CIPCLOSE=1(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[ESP8266 Driver] Channel is not open.  Dropping data
(\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,19:{"getStatus":null}(\r)[ESP8266 Driver] Channel is not open.  Dropping data
1,CLOSED(\r)(\n)[ESP8266 Driver] Socket closed on channel 1
[at] Command timed out
[esp8266] close_cb failed.
[serial (Aux/Wifi)] TX: AT+CIPSEND=0,100(\r)(\n)
[serial (Aux/Wifi)] RX: 1,CONNECT(\r)(\n)[ESP8266 Driver] Socket connected on channel 1
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,16:{"getVer":null}(\r)1,CLOSED(\r)(\n)[ESP8266 Driver] Socket closed on channel 1
[at] Command timed out
[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 0
[wifi] New external connection: Wifi Chan 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[serial (Aux/Wifi)] TX: AT+CIPCLOSE=0(\r)(\n)
[serial (Aux/Wifi)] RX: 1,CONNECT(\r)(\n)[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,16:{"getVer":null}(\r)[at] Command timed out
[esp8266] close_cb failed.
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,191(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: busy s...(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 428 bytes(\r)(\n)
[serial (Aux/Wifi)] RX: SEND FAIL(\r)(\n)[esp8266] send_data_cb failed with msg: Bad response value
[ESP8266 Driver] Failed to send data on channel 1
[wifi] Closing external connection: Wifi Chan 1
[wifi] Stopping telem stream on Wifi Chan 1
[serial (Aux/Wifi)] TX: AT+CIPCLOSE=1(\r)(\n)
[serial (Aux/Wifi)] RX: 1,CLOSED(\r)(\n)[ESP8266 Driver] Socket closed on channel 1
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] RX: 1,CONNECT(\r)(\n)[ESP8266 Driver] Socket connected on channel 1
[wifi] New external connection: Wifi Chan 1
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: +IPD,1,16:{"getVer":null}(\r)
[serial (Aux/Wifi)] TX: AT+CIPSEND=1,191(\r)(\n)
[serial (Aux/Wifi)] RX: (\r)(\n)
[serial (Aux/Wifi)] RX: OK(\r)(\n)
[serial (Aux/Wifi)] TX: {"ver":{"name":"RCP_MK2","fname":"RaceCapture/Pro MK2","major":2,"minor":10,"bugfix":0,"serial":"35004B001851343033393930","git_info":"2.10.0-RC3-17-g8283fa5","release_type":"development"}}(\r)(\n)
[serial (Aux/Wifi)] RX: > (\r)(\n)
[serial (Aux/Wifi)] RX: Recv 191 bytes(\r)(\n)
```

Notice the flood of IPD commands that happens when the new connection opens.  Also take into account that the timeout on a send command for a packet it 5 seconds.  Combined with the message `[serial (Aux/Wifi)] RX: busy s...(\r)(\n)` we see that the ESP8266 gets absolutely swamped with data and does what it can to handle it, but the slow responses cause channel 0 to close (the beacon channel) and in turn since we never checked it (because I figured it would never die because UDP) it never got restarted.  So the fix on my end is to check the beacon channel and re-open it if it is detected to be closed.  On the app side we need to not flood the ESP8266 upon connecting, else this behavior will persist.  Filed https://github.com/autosportlabs/RaceCapture_App/issues/1258 on the App issue tracker

stieg, 2016-09-21:
Should be fixed now @ryandoherty

closed 2016-09-21 by stieg