autosportlabs/RaceCapture-Pro_firmware #868: Battery voltage showing 0.0 in 2.10.0-RC3 or 2.10.0

brentpicasso, 2016-09-30:
2.10.0 release does not show battery voltage.  When falling back to RC2, battery voltage does show in telemetry stream.

STR: 
Have 2.10.0_RC2 firmware programmed. observe battery voltage channel working correctly.
flash 2.10.0-RC3 or 2.10.0 official. 
Note battery voltage channel no longer working.

stieg, 2016-09-30:
Fun.  MK2 or RCT or both?

brentpicasso, 2016-10-03:
Only observed on RCT; RCP with 2.10.0 shows battery voltage correctly.

stieg, 2016-10-05:
Regression is caused by either commit 7f8d5fda2 or commit ef72d3981d (not sure which since the latter leads to a panic).  Either way this is strange behavior since neither of the commits muddle with any of the DMA bits or the ADC settings.  Something is going where it shouldn't

closed 2016-10-12 by stieg