autosportlabs/RaceCapture-Pro_firmware #870: Optimize scheduling of OBDII PIDs of different sample rates

brentpicasso, 2016-10-02:
We should query OBDII PIDs in an optimized way so that faster channels are queried proportionately more often than slower channels.

e.g. RPM @ 50Hz and fuel level @ 1Hz. 

Idea:

maintain a counter for each channel that counts down on every round robin. the channels that count down sooner will 'fire' and be scheduled for querying

mumchristmas, 2016-10-10:
Excuse me, how will the RCT handle the sample rate mismatch (like the e.g.) with 2.10.0 firmware?
Should I set every channel with same rate now?

stieg, 2016-10-11:
@mumchristmas  Internally the unit stores the last sampled value and that is what is returned upon sampling.  So varying sample rates will not make too much of a difference here.  The OBD-II code does round robin queries as fast as it can, thus updating its samples as fast as it can.

meikeric, 2016-11-15:
Please take a look at PR:https://github.com/autosportlabs/RaceCapture-Pro_firmware/pull/892 and let me know your thoughts. I tried to keep it tight, fast and in the loop.

closed 2017-03-10 by brentpicasso