autosportlabs/RaceCapture-Pro_firmware #872: malformed status JSON when cellular is enabled

brentpicasso, 2016-10-03:
STR:

2.10.0 RCP
resetConfig
enable cellular
restart unit
connect app
observe messages in log:
'{"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"57003E000351343038383337","git_info":"2.10.0","release_type":"official","uptime":9337},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":1,"IMEI":"355922062181330' | Unterminated string starting at: line 1 column 280 (char 279)

stieg, 2016-10-03:
So this message just gets printed to the log?  Any other information around where the message is printed?

brentpicasso, 2016-10-03:
This gets printed to the app as a response to the status log query as soon as cellular is enabled and the unit is power cycled.

stieg, 2016-10-20:
An example of the bug:

```
[DEBUG  ] [RCPAPI      ] Tx: {"getStatus":null}
[DEBUG  ] [RCPAPI      ] Rx: {"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"2A0039001347333231343238","git_info":"2.10.0-8-g3665ac5","release_type":"development","uptime":19091},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"","sig_str":0,"number":"","state":"unknown"},"telemetry":{"status":0,"state":"idle","dur":0},"bt":{"init":2},"logging":{"status":0,"dur":0},"track":{"status":0,"valid":false,"trackId":0,"inLap":0,"armed":0},"imu":{"init":true},"wifi":{"active":false,"initialized":false,"ap":{"active":false},"client":{"active":false,"connected":false}}}}
[DEBUG  ] [RCPAPI      ] Tx: {"getStatus":null}
[DEBUG  ] [RCPAPI      ] Rx: {"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"2A0039001347333231343238","git_info":"2.10.0-8-g3665ac5","release_type":"development","uptime":20092},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"","sig_str":0,"number":"","state":"unknown"},"telemetry":{"status":0,"state":"idle","dur":0},"bt":{"init":2},"logging":{"status":0,"dur":0},"track":{"status":0,"valid":false,"trackId":0,"inLap":0,"armed":0},"imu":{"init":true},"wifi":{"active":false,"initialized":false,"ap":{"active":false},"client":{"active":false,"connected":false}}}}
[DEBUG  ] [RCPAPI      ] Tx: {"getStatus":null}
[DEBUG  ] [RCPAPI      ] Rx: {"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"2A0039001347333231343238","git_info":"2.10.0-8-g3665ac5","release_type":"development","uptime":21093},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"","sig_str":0,"number":"","state":"unknown"},"telemetry":{"status":0,"state":"idle","dur":0},"bt":{"init":2},"logging":{"status":0,"dur":0},"track":{"status":0,"valid":false,"trackId":0,"inLap":0,"armed":0},"imu":{"init":true},"wifi":{"active":false,"initialized":false,"ap":{"active":false},"client":{"active":false,"connected":false}}}}
[DEBUG  ] [RCPAPI      ] Tx: {"getStatus":null}
[WARNING] [RCPAPI      ] Message rx worker exception: u'{"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"2A0039001347333231343238","git_info":"2.10.0-8-g3665ac5","release_type":"development","uptime":22095},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"013949008838073' | Unterminated string starting at: line 1 column 295 (char 294)
[DEBUG  ] [Traceback (most recent call last)]
  File "/home/stieg/Devel/asl/app/autosportlabs/racecapture/api/rcpapi.py", line 210, in msg_rx_worker
    msgJson = json.loads(msg, strict=False)
  File "/usr/lib64/python2.7/json/__init__.py", line 352, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib64/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib64/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Unterminated string starting at: line 1 column 295 (char 294)

[WARNING] [RCPAPI      ] Message rx worker exception: u'","sig_str":0,"number":"16088861789","state":"unknown"},"telemetry":{"status":0,"state":"idle","dur":0},"bt":{"init":2},"logging":{"status":0,"dur":0},"track":{"status":0,"valid":false,"trackId":0,"inLap":0,"armed":0},"imu":{"init":true},"wifi":{"active":false,"initialized":false,"ap":{"active":false},"client":{"active":false,"connected":false}}}}' | Extra data: line 1 column 4 - line 1 column 350 (char 3 - 349)
[DEBUG  ] [Traceback (most recent call last)]
  File "/home/stieg/Devel/asl/app/autosportlabs/racecapture/api/rcpapi.py", line 210, in msg_rx_worker
    msgJson = json.loads(msg, strict=False)
  File "/usr/lib64/python2.7/json/__init__.py", line 352, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib64/python2.7/json/decoder.py", line 367, in decode
    raise ValueError(errmsg("Extra data", s, end, len(s)))
ValueError: Extra data: line 1 column 4 - line 1 column 350 (char 3 - 349)
```

stieg, 2016-10-20:
Man... lots of fun here.  Found the bug for the IMEI.  There is a `\r\n` in the string.  Need to strip it out.  

```
{"status":{"system":{"model":"RaceCapture/Pro MK2","ver_major":2,"ver_minor":10,"ver_bugfix":0,"serial":"2A0039001347333231343238","git_info":"2.10.0-8-g3665ac5","release_type":"development","uptime
":124017},"GPS":{"init":1,"qual":0,"lat":0.0,"lon":0.0,"sats":0,"DOP":0},"cell":{"init":0,"IMEI":"013949008838073
","sig_str":0,"number":"16088861789","state":"unknown"},"telemetry":{"status":7,"state":"registration_failed","dur":0},"bt":{"init":2},"logging":{"status":0,"dur":0},"track":{"status":0,"valid":fals
e,"trackId":0,"inLap":0,"armed":0},"imu":{"init":true},"wifi":{"active":false,"initialized":false,"ap":{"active":false},"client":{"active":false,"connected":false}}}}
```

closed 2016-10-21 by stieg