autosportlabs/RaceCapture-Pro_firmware #874: Altitude go 140911008 when below 0

mumchristmas, 2016-10-12:
I live in Shanghai. Some points of the city are below the sea level. :)

stieg, 2016-10-12:
Oops... that is no good.  Thanks for reporting.

stieg, 2016-10-12:
Hey @mumchristmas Do you have a log file perhaps that you can upload that shows this?  Also what version of the firmware are you running?  TIA.

mumchristmas, 2016-10-12:
[RCT-datapoint-161012-ColtMa.xlsx](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/525198/RCT-datapoint-161012-ColtMa.xlsx)

This is what dig form the sq3 file. See the line 7097 to 7539.
My RCT is running official 2.10.0 fw.

stieg, 2016-10-12:
Huh... well this is pretty funny.  Below is the datasheet from our GPS chip vendor:

[image]

We make use of the mean sea level altitude calculation.  Notice how according to the docs this is a UINT32.  Ha!  Wonder if this is a typo in the document or if its a design oversight.  Will reach out to our vendor.

stieg, 2016-10-13:
Looks like documents were wrong, hence the bug:

[image]

Will update code accordingly.

closed 2016-10-13 by stieg