autosportlabs/RaceCapture-Pro_firmware #877: Magnetometer part in imu is not include in the code.

mumchristmas, 2016-10-15:
Trying to do something sensor fusion work to avoid the temperature drift of the imu. And find the compass part is missing in the src. Will you add it in the future? It's an importent part in the fusion calculation.

stieg, 2016-10-17:
We have talked about it but it has not become a super high priority just yet since we haven't found a proper need for it.  What is the fusion calculation you are referring to exactly?  I could see better reason for adding in the temperature channel.

stieg, 2016-10-17:
Also... of course patches are always welcome too :smiley:

mumchristmas, 2016-10-17:
I'm reading some new papers on the data fusion field. And making some research with R language. Not sure which one I will use finally, give me some time. :)

I'm willing to commit codes. But the biggest question is I'm not a ARM guy. All I know about stm32 is from Arduino, LOL.

stieg, 2016-10-17:
> I'm willing to commit codes. But the biggest question is I'm not a ARM guy. All I know about stm32 is from Arduino, LOL.

Eh... its really not very different.  Just a bit less abstraction.  I have docs I can email as needed.

brentpicasso, 2016-12-12:
The MPU9250 we use supports the magnetometer, just to confirm. 

We have this implemented in a branch or two, and I thought it was merged, but not exposed to the app, but checking the current firmware I don't see the magnetometer being queried in I2C. 

https://github.com/autosportlabs/RaceCapture-Pro_firmware/branches/all?utf8=%E2%9C%93&query=compass

Probably should revive the appropriate branch and get this implemented soon, since it's part of the KS campaign.  The I2C implementation was pretty gnarly at the time.

mumchristmas, 2016-12-12:
Thanks Brent. Compass is really useful in the YawRate calibration. Good job!

brentpicasso, 2019-09-03:
Not choosing to implement magnetometer; with vehicle applications there is too much interference from the body, reducing it's usefulness.  We will calculate heading from GPS data.

closed 2019-09-03 by brentpicasso