autosportlabs/RaceCapture-Pro_firmware #878: always report back configured trackId if track database is not present

brentpicasso, 2016-10-17:
In cases when a track database is not present (size == 0) as in RCT,  the meaning of the auto configure flag sould mean that the current track settings should be manipulated externally -it's a flag for the external app to use.   When we do have a track database, as in RCP, the flag should be used internally to manage itself.

To support this, the currently configured trackId should always be reported regardless of the auto configuration setting, *if* the track database is not present. 

Also, we should normalize configuration to always enable track auto-configuration by default.

stieg, 2016-10-17:
So just to clarify here @brentpicasso, if there is a track DB then this flag has different meaning?  If so I hope you understand how complex that is for both user and developer.  Having a flag have dual meaning based on the existence (or lack thereof) of a product feature is a recipe for utter confusion.  Now if the flag has its meaning extended that is consistent across platforms, then this is different.  This is akin to what you and I discussed via 1-1.  I do think its unwise for the firmware to special case behavior based on whether or not it has track information locally.

brentpicasso, 2016-10-17:
It has a consistent meaning insofar as it means "auto configuration should occur" .  Documentation could support that if the device has a track db, then it will auto-configure on it's own; if not, it should be auto-configured externally. 

IMO it is a reasonably-understandable rule handled with documentation, and accommodates the different platforms. It also lets us actively use that field rather than a case where the API field is present, but with no behavior.

stieg, 2016-10-17:
Why not just simplify the entire flow for auto track config to the following procedure:

* If auto-detect enabled on firmware attempt to auto-detect track first using internal track DB.
* If detection of track fails using internal DB, then app will attempt to auto configure track (if app is connected).

This method plays nicely across both platforms without being as confusing and gets roughly the same result.  In the case of RCT the first step will always fails since there is no track DB, thus step two will kick in (since app will see auto-detect is set and no track was set).  In case of MK2 we try to use internal DB to detect track (assuming its up to date) and failing that the app will do the auto-detection (if it is connected).  This is consistent behavior across both platforms this way.  Perhaps this is what you are describing and we just aren't reading it the same?

brentpicasso, 2016-10-18:
So the app does exactly that with RCT; there just happens to be no track DB in this case. On RCT that app directly writes to the track cfg object if auto config flag is true.

brentpicasso, 2016-10-20:
Updated to have MK2 fall back to the fixed configuration if no tracks are auto-detected.  In which case, the app may update the fixed config as it sees fit.  

This is the most consistent behavior that is common between RCT and RCP, mutually agreed upon with @stieg

stieg, 2016-10-20:
Closing issue since merged.

PR #879 (2016-10-20): updates to track autodetection semantics
This PR affects RCT:
* always report configured track ID if we have no track database
* normalizes default configuration.
* Allows auto configure flag to have a useful meaning in RCT, where currently it's a flag that's present but ignored.

This is to match updates to app to enhance track auto-detection, in PR: https://github.com/autosportlabs/RaceCapture_App/pull/1278

Resolves #878
closed 2016-10-20 by stieg