autosportlabs/RaceCapture-Pro_firmware #908: CMSIS While Loop warning

jlwall, 2017-03-02:
Trying to compile r/2.11.0  with 'make mk3 or make mk2' I get the following build error

Either something in the make setting need to change to allow warnings to pass, or the CMSIS driver needs a patch

> In function 'SetSysClock':
> libs/STM32F4xx_DSP_StdPeriph_Lib_V1.0.1/Libraries/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c:408:9: error: this 'while' clause does not guard... [-Werror=misleading-indentation]
>          while ((RCC->CFGR & (uint32_t)RCC_CFGR_SWS ) != RCC_CFGR_SWS_PLL);
>          ^~~~~
> libs/STM32F4xx_DSP_StdPeriph_Lib_V1.0.1/Libraries/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c:409:9: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'while'
>          {
>          ^
> cc1: all warnings being treated as errors

L408 is
 while ((RCC->CFGR & (uint32_t)RCC_CFGR_SWS ) != RCC_CFGR_SWS_PLL);
{
}

Changing this to removes the warning which allows the build to complete
 while ((RCC->CFGR & (uint32_t)RCC_CFGR_SWS ) != RCC_CFGR_SWS_PLL)
{
}

brentpicasso, 2017-03-02:
Hi, what does arm-none-eabi-gcc -v output?  

This is what we have:

gcc version 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601] (GNU Tools for ARM Embedded Processors)

jlwall, 2017-03-06:
gcc version 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739] (GNU Tools for ARM Embedded Processors) 

Sorry I think this got updated from a different project, will roll back to 4.7.4

closed 2017-03-06 by jlwall