autosportlabs/RaceCapture-Pro_firmware #909: firmware compile error

mach1000, 2017-03-04:
Hi,
I have made local copy of the firmware on a fresh ubuntu 16.10 and followed the the requirements for building firmware.
When I compile the firmware with "make mk2" it fails at loggerapi.c with the following error.

**************************''

../../src/logger/loggerApi.c: In function 'rc_version_info':
../../src/logger/loggerApi.c:123:47: error: expected expression before ',' token
         json_int(serial, major_name, MAJOR_REV, 1);
                                                                            ^
../../src/logger/loggerApi.c:124:47: error: expected expression before ',' token
         json_int(serial, minor_name, MINOR_REV, 1);
                                                                            ^
../../src/logger/loggerApi.c:125:45: error: expected expression before ',' token
         json_int(serial, bf_name, BUGFIX_REV, 1);
                                                                       ^
Makefile:134: recipe for target '../../src/logger/loggerApi.o' failed
make[1]: *** [../../src/logger/loggerApi.o] Error 1
make[1]: Leaving directory '/home/rcp/RaceCapture-Pro_firmware-master/platform/mk2'
Makefile:110: recipe for target 'mk2-build' failed
make: *** [mk2-build] Error 2

brentpicasso, 2017-03-04:
Hi, what branch are you compiling, and also what does gcc -v output for you?

mach1000, 2017-03-05:
Hi Brent, my 16.10 just crashed on me second time, so I will get 16.04 instead.

I was compiling latest firmware, I downloaded it as a zip, but will do it the git way when I get things running.

I will get the enviroment up and running and then get back to this thread if I have same issue,

mach1000, 2017-03-05:
just got a 16.04 up and running and installed dependencies but still fails at same point.

added path to gcc-arm..... in  ~/.bashrc

gcc -v

Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4)

mach1000, 2017-03-06:
Hi guys, I am using the vagrant image now so no need to try and find the error pasted in. probably is caused by my self

brentpicasso, 2017-03-06:
Ok, thanks for the update. Closing the issue for now.

closed 2017-03-06 by brentpicasso