autosportlabs/RaceCapture-Pro_firmware #932: Heading and magnetic compass support

brentpicasso, 2017-06-28:
Enable heading calculated by GPS, and also enable the MPU-9150 and MPU-9250 magnetic compass. 

Need to confirm solution works with MPU-9150 as well as the MPU-9250 found in later revisions of MK2. 

Proof of concept implementation can be found here; would need to integrate into latest branch. 
https://github.com/autosportlabs/RaceCapture-Pro_firmware/tree/v2.9.0_newmod_mag_gps_heading

kktse, 2019-01-19:
Although I am primarily interested in the use case in Issue #997, this is an interesting enhancement that would help provide more data to resolve the necessary information to build a filter. Specifically, I am interested in being able to resolve the full velocity vector from the GPS speed.

I noticed the calculation for heading uses a backwards difference to measure the change in position.
https://github.com/autosportlabs/RaceCapture-Pro_firmware/blob/4ae084be680b0ff5f04e08c164d95cc497a83a75/src/gps/geopoint.c#L48-L67

Browsing some literature related to #997, it seems that GPS velocities is the measurement used for state estimation. In that view, would it make more sense to provide an instantaneous heading by taking the arctangent of the velocity vectors? This is a more direct calculation that forgoes the backwards difference. I've written some code that exposes the GPS velocities which might help. 

As for the magnetometer, I have a pro/mk2 which I can help test the magnetometer code.

brentpicasso, 2019-09-03:
Not choosing to implement magnetometer; with vehicle applications there is too much interference from the body, reducing it's usefulness.  We will calculate heading from GPS data.

closed 2019-09-03 by brentpicasso