autosportlabs/RaceCapture-Pro_firmware #997: GPS + IMU fusion frame without magnetometer

dgala, 2018-06-21:
I am trying to implement the Kalman filter for GPS + IMU fusion. I have the position and velocity (at a low rate) from the GPS module. I further have the accelerometer and gyroscope output in the IMU frame. Can anyone help me with this implementation without using the magnetometer?

brentpicasso, 2018-06-21:
Hi, thanks for your interest.  What exactly do you have working right now, and what exactly would you like the results to be? 

The easiest way to integrate a Kalman filter into the system is to apply a correction to the existing GPS Latitude, Longitude and Speed channels, which will be transparent to external systems; it would just benefit from the corrected / filtered data.

dgala, 2018-06-21:
Hi brentpicasso, thanks for replying. I currently have the GPS positions and velocities in ECEF frame. I also have the accelerations in IMU frame. I am trying to implement an EKF with positions, velocities, attitude, biases in attitude and biases in accelerations as states. I understand that I also need to address the gravity in here.

brentpicasso, 2019-09-03:
Closing due to no updates.

closed 2019-09-03 by brentpicasso