autosportlabs/RaceCapture-Pro_firmware #1003: do_release.sh errors out with -Werror=misleading-indentation

eacmen, 2018-09-16:
I am trying to test/submit a PR and following the contributing guide `do_release.sh` must pass. However this script does not pass even on clean release tags I have checked out. It always errors on `-Werror=misleading-indentation`, which I assume is trying to enforce a coding standard? Here is the output:
Relavant info:
```
EV=1 -DMAJOR_REV=2 -DMINOR_REV=13 -DBUGFIX_REV=5 -DRC_BUILD_GIT_DESCRIPTION="\"2.13.5\"" -DRC_BUILD_RELEASE_TYPE=RELEASE_TYPE_OFFICIAL   -O0 -DRCP_TESTING -DAPI_REV=1 -DMAJOR_REV=2 -DMINOR_REV=13 -DBUGFIX_REV=5 -DRC_BUILD_GIT_DESCRIPTION="\"2.13.5\"" -DRC_BUILD_RELEASE_TYPE=RELEASE_TYPE_OFFICIAL  -I. -I./include -I.. -I../logger -I../platform/mk2/hal/fat_sd_stm32/fatfs -I../include -I../include/logging -I../include/gps -I../include/gsm -I../include/jsmn -I../include/api -I../include/filter -I../include/logger -I../include/channels -I../include/virtual_channel -I../include/tracks -I../include/tasks -I../include/messaging -I../include/modem -I../include/lua -I../include/command -I../include/predictive_timer -I../include/auto_config -I../include/util -I../include/serial -I../include/devices -I../include/drivers -I../include -I../include/usart -I../include/usb_comm -I../include/imu -I../include/ADC -I../include/timer -I../include/PWM -I../include/LED -I../include/cpu -I../include/CAN -I../include/OBD2 -I../include/GPIO -I../include/watchdog -I../include/memory -I../include/sdcard -I../include/system -I../include/lap_stats -I../include/units -Ilogger_mock -Igps -Ican_obd2 -Ilap_stats -IFreeRTOS_Kernel/ -IFreeRTOS_Kernel/include -IFreeRTOS_Kernel/include_testing -Iutil -I../src -I../src/devices -I../src/lap_stats -I../src/logger -I../src/modem  -D_RCP_BASE_FILE_="\"can_channels.c: \"" -c ../src/CAN/can_channels.c -o build/rcp_base/src/CAN/can_channels.o
../src/CAN/can_mapping.c: In function ‘canmapping_map_value’:
../src/CAN/can_mapping.c:108:9: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
         if (! canmapping_match_id(can_msg, mapping))
         ^~
../src/CAN/can_mapping.c:111:3: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
   uint64_t raw_data = can_msg->data64;
   ^~~~~~~~
cc1: all warnings being treated as errors
Makefile:130: recipe for target 'build/rcp_base/src/CAN/can_mapping.o' failed
make[3]: *** [build/rcp_base/src/CAN/can_mapping.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/workspace/m3/RaceCapture-Pro_firmware_og/test'
Makefile:73: recipe for target 'test-build' failed
make[2]: *** [test-build] Error 2
make[2]: Leaving directory '/workspace/m3/RaceCapture-Pro_firmware_og'
Makefile:81: recipe for target 'test-pristine' failed
make[1]: *** [test-pristine] Error 2
make[1]: Leaving directory '/workspace/m3/RaceCapture-Pro_firmware_og'
Makefile:234: recipe for target 'package' failed
make: *** [package] Error 2
```

When I build just the mk3 firmware I am interested in it builds just fine.

brentpicasso, 2018-09-16:
Hi, just double-checked the build_release script against master, and it built successfully.

Can you confirm the OS and version of GCC you're using?

eacmen, 2018-09-16:
I am using `gcc-arm-none-eabi-4_9-2014q4` download linked off the main page:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Host native compiler:
```
$ gcc --version
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I assumed the whole build was using the arm compiler. but it looks like your test harness is building the code natively? This is likely the issue.

I am running debian 9. I tried configuring my system gcc to be 6.0, 4.9 and 4.6 and they all error'd out in different ways. What gcc version is required for the native build?

brentpicasso, 2018-09-16:
Hi, 

this is what gcc --version reports currenty:

gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609

brentpicasso, 2018-10-08:
Closing due to inactivity. Peter, if you want to just submit the PR, we can take a look at it.  Thanks!

closed 2018-10-08 by brentpicasso