autosportlabs/RaceCapture-Pro_firmware #1022: When logging at 25hz, 10hz are drop to 5hz

kokarez, 2019-04-18:
When logging at 25hz, 10hz are drop to 5hz. This is due to tick count of 25hz not overlapping with the 10hz, thus half of the ticks will be drop by the should_sample which for all sample to be align on 25hz.

brentpicasso, 2019-05-10:
Thanks for the note. Can you attach an example config that reproduces this problem?

xavierd, 2019-11-10:
I've also just seen this. The data logged for configured 10Hz sensors is actually only logged at 5Hz, including the GPS data :(. See the attached configuration.
[conf.txt](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/3829042/conf.txt)

xavierd, 2019-11-10:
And this is a log generated with the config: [rc_2.log](https://github.com/autosportlabs/RaceCapture-Pro_firmware/files/3829046/rc_2.log). If you take a look at the OilPressure for instance, you can see that a value is generated every 0.2s (ie 5Hz), while being configured for 10Hz.

brentpicasso, 2020-06-21:
You can work-around this situation by adding a channel that logs at 50Hz or higher, or changing the Accel/Gyro channel to log at 10Hz or 50Hz.  We will work to get this resolved in an upcoming release.

closed 2020-07-04 by brentpicasso