autosportlabs/RaceCapture-Pro_firmware #1031: getChannel() returning nil

zerox1212, 2019-05-10:
When a user calls `getChannel()` on a valid channel `nil` should be the exception, not the norm.

This is essentially a high level function and as a user I expect the last available data to be here regardless of channel update rate and `tick()` settings.

The current implementation, with the right combination of channel update rate and tick, seems to make it possible to get almost nothing but `nil` returned.

Bare minimum the docs should describe that calling `getChannel()` faster than the update will result in a lot of `nil` getting returned.

brentpicasso, 2019-05-11:
Thank you for the note.  Documentation has been clarified in the wiki: https://wiki.autosportlabs.com/RaceCapturePro_Lua_Scripting#getChannel.28_Channel_ID_or_name_.29

closed 2019-05-11 by brentpicasso

zerox1212, 2019-05-11:
Thanks for the update.

Can you confirm that if a channel's rate is set to 1hz, and the script tick is set to 1hz, that desync can make it so that you only get `nil`?

I did a test and it seems like if update rate and tick are set to `1` it could take several minutes before you get a number instead of `nil`.