autosportlabs/RaceCapture-Pro_firmware #1106: txCAN truncates extended CAN ID

brentpicasso, 2021-07-21:
txCAN seems to truncate
"Problem is that data is sent to address 0x06040000 instead of 0x0607ff80"

```
local setvolt400 = {3,0,0,33,67,200,0,0} 
local setvolt420 = {3,0,0,33,67,210,0,0}
local setcurrent100 = {3,0,0,34,64,96,0,0}
local setcurrent45 = {3,0,0,34,63,192,0,0}
local seton = {3,0,0,48,0,0,0,0}
local setoff = {3,0,0,48,0,1,0,0}
local allmodules = 0x0607ff80

function onTick()

press = getGpio(1)                      --charge switch status
pv=getChannel("PackVolts")
if press == 0 and pv < 405 then
  txCAN(0,allmodules,1,setvolt420)
  txCAN(0,allmodules,1,setcurrent100)
  txCAN(0,allmodules,1,seton)
--  println("on")
else 
   txCAN(0,0x0607ff80,1,setoff)
-- println("off")
end

end```

RodADavison, 2021-07-24:
With CAN1&CAN2 looped together, this script does not indicate any error with these values:

local seton = {3,0,0,48,0,0,0,0}
local allmodules = 0x0607ff80
setTickRate(30)

function onTick()
    txCAN(0,allmodules,1,seton)
    id, ext, data = rxCAN(1)
    while id ~= nil do
       println("CAN rx: " ..id .." " ..ext.. " " ..data[1])
       id, ext, data = rxCAN(1)
    end
end

RodADavison, 2021-08-01:
This was determinted to be external to the rcp.

closed 2021-08-01 by RodADavison